nix-community/raspberry-pi-nix #2: narHash for two inputs is wrong

lilyball, 2023-05-30:
I'm seeing two incorrect narHashes in your flake.lock. I had to work around this by manually "updating" each input back to the same revision. In both cases the `lastModified` and `rev` stayed the same, but the `narHash` changed. This was with Nix 2.13.3.

* `rpi-firmware-stable-src` - the hash I see is `sha256-00XrEDj2n3627MJ75waI3M0t40MFevFgGOnpHjjP0dQ=`
* `rpi-linux-5_15-src` - the hash I see is `sha256-/5UStX6DUxWQ2WFWvB5u1TszLUzCo+b/qiBVetEkBTE=`

I have no explanation for why the hashes are wrong, unless github actually changed the archive file it returns for the locked revisions (which would be a rather big surprise).

tstat, 2023-05-30:
I am seeing the same hash as is currently in the lockfile when I removed `rpi-linux-5_15-src` from my nix store and updated with `nix flake lock --recreate-lock-file`. Are you on macos? I am wondering if this could this be something like: https://github.com/NixOS/nix/issues/6731.

lilyball, 2023-06-01:
@tstat I am indeed on aarch64-darwin.

lilyball, 2023-06-09:
I'm going to close this. The underlying issue is a case collision and the fact that Nix doesn't seem to apply `use-case-hack` to flake inputs. I'm currently looking into whether it's feasible to switch my store to a case-sensitive one.

closed 2023-06-09 by lilyball

lilyball, 2023-06-10:
For anyone else who runs into this, the solution was not just to migrate my store to case-sensitive (which was very nerve-wracking given how much of my system depends on it), but also to make a `/nix/tmp` directory and set `TMPDIR=/nix/tmp` when running `nix`, as it apparently downloads flake inputs into the tmpdir first before copying them to the store.

LoganBarnett, 2024-06-03:
@lilyball thanks for sharing the workaround!  To confirm - did you follow something akin to this? https://github.com/cr/MacCaseSensitiveConversion
Summarized: time machine backup, change backup's sensitivity flag, time machine restore.

And if so, have you been fairly happy with the outcome of that?   I also find the prospect nerve wracking :D

I also went down the rabbit hole of trying to perform the build directly on the `linux-builder`, but I can't get it to simply run `nix` - it's like a very partial NixOS setup that I haven't figured out how to expand.

lilyball, 2024-06-03:
@LoganBarnett No, I did not do anything with backups. IIRC I created a new case-sensitive volume, then copied the entire store onto it, then grepped for any files with the case sensitive hack and asked nix to delete those store paths, and also asked nix to delete the store paths for the broken flake inputs (thankfully my running system did not depend on any case-hacked files). If I had to do it again I'd probably ask nix to copy my current profiles to the other store instead of doing a filesystem copy. I also did the swap between stores in single-user mode.

As for the tmpdir, I overrode `config.nix.package` to point to a `buildEnv` that wraps `nix`, then delete the `$out/bin` symlink and makes a folder there that uses `makeBinaryWrapper` to wrap every nix binary to set `TMPDIR=/nix/tmp`. This way every invocation of a nix binary will use a tmpdir on the nix store. I also had to set the permissions of `/nix/tmp` to the same as `/private/tmp`. This all works, although as of macOS 14 there's no way to tell the system to include `/nix/tmp` in the daily tempfile cleaning (but it looks like nix generally does a good job cleaning up after its temporary files).