nix-community/raspberry-pi-nix #3: Update for Linux 6.1 kernel

lilyball, 2023-06-10:
Is it easy to update this for the Linux 6.1 kernel? NixOS 23.05 switched to that as the default, and the upstream raspberrypi/linux repo now defaults to the rpi-6.1.y branch.

tstat, 2023-06-10:
It should be an easy update. I am in progress on similar changes: I rebuilt my pis with 23.05 yesterday and am upgrading them to the 6.1 kernel to test with today. Assuming that all goes well I'll merge the kernel-6.1 branch into master.

I pushed the 5.15 and 6.1 kernels along with other packages from the overlay built against nixos-23.05 into cachix, so hopefully that saves you some time.

tstat, 2023-06-12:
https://github.com/tstat/raspberry-pi-nix/pull/5

closed 2023-06-12 by tstat