nix-community/raspberry-pi-nix #6: Bluetooth: hci0: BCM: Read controller features failed (-110)

zeronobody0, 2023-07-03:
I built an [image using this flake](https://codeberg.org/shitpostalotl/nixos/src/branch/main/hosts/pi.nix), however bluetooth does not work. See issue title as well as [full dmesg output](http://0x0.st/H1AS.txt).

tstat, 2023-07-03:
At a glance I don't see what is wrong with your config. 

When you say you built an image, do you mean that you built an sd card image with `nix build '.#nixosConfigurations.rpi-example.config.system.build.sdImage'` and flashed it to an sd card?

zeronobody0, 2023-07-03:
building: [yeah i think so](https://codeberg.org/shitpostalotl/nixos/src/commit/4332dde0e57d444f9d70dc515397f725fd27bb86/flake.nix#L40)
flashing: yes

tstat, 2023-07-03:
Can you paste the output of `dtc -I fs /sys/firmware/devicetree/base`?

zeronobody0, 2023-07-03:
That will take a while, as the pi doesn't seem to have networking capabilities, so I must recompile the image to add the dtc packadge.

zeronobody0, 2023-07-03:
[yeah i dont think you want me to paste that directly here it's really long](http://0x0.st/H1mk.txt)

tstat, 2023-07-04:
I see, I inadvertently broke bluetooth with a recent change to get uart kernel output on boot. It should be working now on master, and the latest release doesn't have this problem.

commit 9f536c07 (2023-07-04): Revert "enable uart console for kernel"

This reverts commit 5f94c31db1d2c734523a98ae2b490891a6dc3ac7.

fixes #6

ttyAMA0 controls uart0, which can be used for bluetooth. If it is used
for bluetooth then we don't want to write a bunch of garbage to it.
closed 2023-07-04 by tstat