nix-community/raspberry-pi-nix #11: RPI doesnt boot nixos when Serial hat is on

blazp7, 2024-03-01:
Hello ive tried to use your module to fix this error but it doesnt help.
[image]
Do you have any advice why nixos does not boot when i connect this hat https://abelectronics.co.uk/p/77/rs485-pi but works fine without? I need to read data from /dev/serial device but have been unsuccessful using Nixos..

This is my config
```nix
{inputs, ...}: {
  imports = [inputs.raspberry-pi-nix.nixosModules.raspberry-pi];
  nix.settings = {
    substituters = ["https://raspberry-pi-nix.cachix.org"];
    trusted-public-keys = ["raspberry-pi-nix.cachix.org-1:WmV2rdSangxW0rZjY/tBvBDSaNFQ3DyEQsVw8EvHn9o="];
  };
  hardware = {
    bluetooth.enable = false;
    raspberry-pi.config = {
      all = {
        options = {
          # ive tried a bunch of options here
        };
      };
    };
  };
}
```
Also added substituters in flake.nix, however the rebuild still takes 20+ hours :(
```nix
  nixConfig.extra-substituters = ["https://raspberry-pi-nix.cachix.org"];
  nixConfig.extra-trusted-public-keys = ["raspberry-pi-nix.cachix.org-1:WmV2rdSangxW0rZjY/tBvBDSaNFQ3DyEQsVw8EvHn9o="];
```
Help with serial error much appreciated, thanks.

tstat, 2024-03-05:
Hey blazp7. Sure, I can try to help.

This repo and nixpkgs both use uboot as the bootloader, and uboot has a serial interface outputting the same information you posted in your screenshot and listening for input, so I would expect a loopback to cause uboot to fail.  

Raspbian does not boot with uboot, so it wouldn't have this issue with a loopback. 
 
I'm not sure why connecting the hat is causing issues though. Can you paste your `/boot/firmware/config.txt`?

blazp7, 2024-03-07:
Thankfully I have mostly solved the issue. It was a non-problem, i only needed to connect the wires from the serial extension hat to my device. It seems that just letting the rs485 connector/wires dangle in the air is picking up some interference? If i connect the rs485 connector/wires to my device it boots successfully. I can even just pinch the wires together, it still goes through successfully with the boot process.

I was trying to nix overlay uboot with the silent boot flags (links below) but i was unsuccessful. Maybe it is worth consideration still but for now it works!
https://raspberrypi.stackexchange.com/questions/116074/how-can-i-disable-the-serial-console-on-distributions-that-use-u-boot
https://raspberrypi.stackexchange.com/questions/116074/how-can-i-disable-the-serial-console-on-distributions-that-use-u-boot

closed 2024-03-07 by blazp7