nix-community/raspberry-pi-nix #13: Pi 5 support

adminy, 2024-04-18:
Awesome project! As you know some things are going out of nixpkgs repo for 24.11, and official support for the pi 5 won't be prioritised. So I'm looking forward to see that support here!

Thanks

tstat, 2024-05-02:
Thanks! Raspberry pi support works as of b39b556e8a1c4bd6f8a59e8719dc1f658f18f255, just not with uboot, so you need to disable it with `raspberry-pi-nix.uboot.enable = false` ([option documented here](https://github.com/tstat/raspberry-pi-nix/blob/master/rpi/default.nix#L30-L43)).

adminy, 2024-05-07:
wifi seems to not be working.

malteneuss, 2024-05-08:
As far as i know [basic pi 5 support](https://patchwork.ozlabs.org/project/uboot/list/?series=391659&state=%2A&archive=both) (e.g. booting from SD but not USB,NVMe) was added in uboot version 2024.04, which has already landed in nixpkgs-unstable https://github.com/NixOS/nixpkgs/blob/8f9b659d9fb9af9d39cef689e2c3da5dd661b23a/pkgs/misc/uboot/default.nix#L31

Would that help here to not need to disable uboot?

tstat, 2024-05-08:
Oh great, I didn't know a new version of uboot was released. I'll probably give it a test later. If you are motivated to then you probably just need to update the source and version in these two places.

https://github.com/tstat/raspberry-pi-nix/blob/master/flake.nix#L7 
https://github.com/tstat/raspberry-pi-nix/blob/master/overlays/default.nix#L46

tstat, 2024-05-08:
@adminy hard to have anything to reply to without more information. Wifi, bluetooth, etc all work fine with my pi 5 and the example project linked in the readme. I've been using it with nixos for months.

malteneuss, 2024-05-08:
@tstat I'll wait for the bump to land in your main branch ;) I'm currently working on the documentation for NixOS on Pi 5 at https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_5 and tinkering with how to use the official Pi 5 Debian OS with Nix as a remote builder to native-build a pre-setup NixOS image using your flake (i couldn't get cross-compile to work, and emulated native-compile is way too slow). The goal is to document the steps and make it easy for people to get NixOS on the Pi until NixOS hardware has better Pi 5 support and cached compiled kernels.

Thinking of which, do you see a way to move this setup to NixOS-hardware for greater visibility? I wasn't able to find your repo until someone mentioned it. And btw, could you explain why we would want uboot when it seems to work without? Other people are pushing for UEFI, because only it would allow the usual systemd-boot bootloader with support for different NixOS generations; as i understood uboot wouldn't support that.

shnCanos, 2024-05-12:
Hello, this might be a bit of a noob question, but how is the option `raspberry-pi-nix.uboot.enable = false` meant to be enabled? With the configuration below (which is [the example repo](https://github.com/tstat/raspberry-pi-nix-example) with a modified `modules` section)
```nix
{
  description = "raspberry-pi-nix example";
  nixConfig = {
    extra-substituters = [ "https://raspberry-pi-nix.cachix.org" ];
    extra-trusted-public-keys = [
      "raspberry-pi-nix.cachix.org-1:WmV2rdSangxW0rZjY/tBvBDSaNFQ3DyEQsVw8EvHn9o="
    ];
  };
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/8bf65f17d8070a0a490daf5f1c784b87ee73982c";
    raspberry-pi-nix.url = "github:tstat/raspberry-pi-nix";
  };

  outputs = { self, nixpkgs, raspberry-pi-nix }:
    let
      inherit (nixpkgs.lib) nixosSystem;
      basic-config = { pkgs, lib, ... }: {
        time.timeZone = "America/New_York";
        users.users.root.initialPassword = "root";
        networking = {
          hostName = "basic-example";
          useDHCP = false;
          interfaces = { wlan0.useDHCP = true; };
        };
        environment.systemPackages = with pkgs; [ bluez bluez-tools ];
        hardware = {
          bluetooth.enable = true;
          raspberry-pi = {
            config = {
              all = {
                base-dt-params = {
                  # enable autoprobing of bluetooth driver
                  # https://github.com/raspberrypi/linux/blob/c8c99191e1419062ac8b668956d19e788865912a/arch/arm/boot/dts/overlays/README#L222-L224
                  krnbt = {
                    enable = true;
                    value = "on";
                  };
                };
              };
            };
          };
        };
      };

    in
    {
      nixosConfigurations = {
        rpi-example = nixosSystem {
          system = "aarch64-linux";
          modules = [ raspberry-pi-nix.nixosModules.raspberry-pi basic-config { raspberry-pi-nix.uboot.enable = false; } ]; # <-- Added option here
        };
      };
    };
}
```
I get an error saying that the option `raspberry-pi-nix.uboot` doesn't exist when I run `nix build --extra-experimental-features "nix-command flakes" '.#nixosConfigurations.rpi-example.config.system.build.sdImage'` as root.

malteneuss, 2024-05-12:
It works on my machine with the function-style NixOS-config for that setting:

```nix
modules = [
  raspberry-pi-nix.nixosModules.raspberry-pi
  ({ pkgs, ... }: {
     # uboot not yet supported for pi5
     # https://github.com/tstat/raspberry-pi-nix/issues/13#issuecomment-2090601812 
     raspberry-pi-nix.uboot.enable = false;
    })
   ...
];
``` 

That shouldn't make a difference, but I can't see any other difference.

shnCanos, 2024-05-12:
Thanks for the quick reply! Nothing changed, however I'll try to investigate it and see if I can figure it out

amorx1, 2024-05-13:
I've got the RPi 5 booting using the provided example, but keep getting the following error on every `nixos-rebuild switch`:

```shell
...
updating GRUB 2 menu ...
Failed to get blkid info (returned 512) for / on /dev/root at /nix/store/___________-install-grub.pl line 208.
warning: error(s) occured while switching to the new configuration
``` 

with a single generated fileSystem in `hardware-configuration.nix`:
```nix
fileSystems."/" = 
    { device = "/dev/disk/by-uuid/4444444-4444-4444-8888-888888888888";
       fsType = "ext4";
    };
```

Has anyone else come across this error? It looks like the derivation builds successfully but fails just as the switch is meant to happen.

tstat, 2024-05-13:
@amorx1 it looks like you have enabled grub. This repo doesn't support that bootloader -- it only supports uboot or raspberry pi's normal boot process.

Note that this repo disables grub.
https://github.com/tstat/raspberry-pi-nix/blob/bc745f7dc1089b2fae2b54cb69e8f0a35bf5a688/rpi/default.nix#L278

tstat, 2024-05-14:
@malteneuss 

> I'm currently working on the documentation for NixOS on Pi 5 at https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_5 and tinkering with how to use the official Pi 5 Debian OS with Nix as a remote [...]

Oh, very cool. I compile the raspberry pi kernel on nixos-running raspberry pis. Note that I have sd images of releases in [the example repo](https://github.com/tstat/raspberry-pi-nix-example).

> Thinking of which, do you see a way to move this setup to NixOS-hardware for greater visibility? (Some generic kernel was introduced recently with https://github.com/NixOS/nixos-hardware/pull/927)

I'm happy to discuss moving this stuff to somewhere more visible. 

> And btw, could you explain why we would want uboot when it seems to work without? Other people are pushing for UEFI, because only it would allow the usual systemd-boot bootloader with support for different NixOS generations; as i understood uboot wouldn't support that.

uboot supports different NixOS generations -- when you boot up the example config for example it presents a list of previous generations to select from if desired. This is presented over hdmi and uart.

tstat, 2024-05-15:
pi 5 works with uboot as well in latest release https://github.com/tstat/raspberry-pi-nix/releases/tag/v0.3.0

closed 2024-05-15 by tstat

malteneuss, 2024-05-16:
@tstat Thanks for the update.

Understanding this post https://github.com/NixOS/nixpkgs/issues/260754#issuecomment-1761957084 a bit better now i think that the nixpkgs and nixos-hardware people don't want (to maintain) Pi-specific kernels, firmware etc, like is being done in this repo. So there's probably no better place to move this repo into. Instead i added it to the Wiki guide at https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_5 to be easier to find in the mean time.

DavHau, 2024-12-13:
Did anyone get this to work on a pi5 with nixos-unstable?
I cannot get my pi5 to boot. I can see the kernel starting, but it gets stuck after a few seconds. It works with nixos stable though.

adminy, 2024-12-14:
yep, it works, just make sure you disable u-boot. I have 10 pis in prod with this @DavHau

spacekitteh, 2025-01-01:
Why does nixos-stable work with uboot but not nixos-unstable?

appaquet, 2025-01-05:
Is this a known issue that the ethernet adapter is not working (latest of this repo + nixpkgs-24.11)? I've seen somewhere else floating that only the WiFi adapter works, but couldn't find anything here. The OS boots fine, but dhpcd can't seem to be able to get an IP, but nothing stands out in kernel messages (which obviously works with Rpi OS). Assigning a static ip doesn't work either. The link stats is showing RX trafic, but nothing in TX.

tstat, 2025-01-08:
> Is this a known issue that the ethernet adapter is not working (latest of this repo + nixpkgs-24.11)? I've seen somewhere else floating that only the WiFi adapter works, but couldn't find anything [...]

Looks like an upstream issue https://github.com/raspberrypi/linux/issues/6420. I experienced this when I upgraded my pi 5 to the 6.6.67 kernel, so I added the latest kernel tagged stable in this PR: https://github.com/nix-community/raspberry-pi-nix/pull/107, and my pi 5 works fine with that kernel.

appaquet, 2025-01-09:
> > Is this a known issue that the ethernet adapter is not working (latest of this repo + nixpkgs-24.11)? I've seen somewhere else floating that only the WiFi adapter works, but couldn't find anything here. The OS boots fine, but dhpcd can't seem to be able to get an IP, but nothing stands out in kernel messages (which obviously works with Rpi OS). Assigning a static ip doesn't work either. The link stats is showing RX trafic, but nothing in TX.
> 
> Looks like an upstream issue [raspberrypi/linux#6420](https://github.com/raspberrypi/linux/issues/6420). I experienced this when I upgraded my pi 5 to the 6.6.67 kernel, so I added the latest kernel tagged stable in this PR: #107, and my pi 5 works fine with that kernel.

Yep, it fixed it. Thanks!

diferhe, 2025-02-19:
> Yep, it fixed it. Thanks!

I have a similar issue with the latest kernel v6_12_11 from PR: #102