nix-community/raspberry-pi-nix #14: `libcamera` build error

PowerUser64, 2024-05-02:
I've been getting an error for the past months, and it turns out it's from this. Pipewire expects libcamera v0.2.0 or later, but raspberry-pi-nix has an overlay that makes it v0.1.0. 

See also: https://discourse.nixos.org/t/44644

```shell
error: builder for '/nix/store/xsdmp9630lbrb329m7wmsig0y90ilxli-pipewire-1.0.5.drv' failed with exit code 1;
       last 10 log lines:
       > Run-time dependency jack found: YES 1.9.22
       > Run-time dependency vulkan found: YES 1.3.280
       > Has header "vulkan/vulkan.h" with dependency vulkan: YES
       > Dependency libcamera found: NO found 0.1.0 but need: '>= 0.2.0'
       > Run-time dependency libcamera found: NO (tried pkgconfig and cmake)
       > Looking for a fallback subproject for the dependency libcamera
       >
       > spa/meson.build:99:18: ERROR: Automatic wrap-based subproject downloading is disabled
       >
       > A full log can be found at /build/source/build/meson-logs/meson-log.txt
       For full logs, run 'nix log /nix/store/xsdmp9630lbrb329m7wmsig0y90ilxli-pipewire-1.0.5.drv'.
error: 1 dependencies of derivation '/nix/store/w8nn0l6f8ld413mj91q54r6iqyifbgsp-SDL2-2.30.2.drv' failed to build
error: 1 dependencies of derivation '/nix/store/9y234ggxyfccjwd3w4ad4ca4rl34ciib-openal-soft-1.23.1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/yq0inaf139nkgdd0hyrxj9h366685lnr-gst-plugins-bad-1.22.9.drv' failed to build
error: 1 dependencies of derivation '/nix/store/kv26al19kx3xjyzdczkn38bm19kgyfal-gtk4-4.12.5.drv' failed to build
error: 1 dependencies of derivation '/nix/store/acj6maaw2q2d9b9202rrbgk50vfxabam-NetworkManager-fortisslvpn-gnome-1.4.0.drv' failed to build
error: 1 dependencies of derivation '/nix/store/dxhrk8vjw6f6ln0ms0wgv3agmqrzhd2w-NetworkManager-l2tp-gnome-1.20.14.drv' failed to build
error: 1 dependencies of derivation '/nix/store/6vrdzir8rsrmjm7mm93c4rh8vrm0qxhw-NetworkManager-openconnect-1.2.10.drv' failed to build
error: 1 dependencies of derivation '/nix/store/1siqm60x7qjy171rl29qf3dm4pnvjbna-NetworkManager-openvpn-1.10.2.drv' failed to build
error: 1 dependencies of derivation '/nix/store/wx8iys17jsjid56i96h12w4y4ghw8wqs-NetworkManager-sstp-gnome-1.3.2.drv' failed to build
error: 1 dependencies of derivation '/nix/store/9d8qaggvzb71vv62w8hssmrn9zkdgqx6-NetworkManager-vpnc-1.2.8.drv' failed to build
error: 1 dependencies of derivation '/nix/store/p5kha7pb2rxshapdi6pc7w00x7s6mgwp-gcr-4.1.0.drv' failed to build
error: 1 dependencies of derivation '/nix/store/idagc605vbacjm4kwkrcf4lmngcr5c2p-etc.drv' failed to build
error (ignored): error: cannot unlink '/tmp/nix-build-linux-6.1.63.drv-1/source/build': Directory not empty
error: 1 dependencies of derivation '/nix/store/rxgx052vn5bq5msh5imxxg77vh53l93w-system-path.drv' failed to build
error: 1 dependencies of derivation '/nix/store/dn47rzzwrzjjmpcrsr4qdr6yvadlk48f-nixos-system-srv-24.05.20240427.58a1abd.drv' failed to build
```

tstat, 2024-05-02:
👋  It sounds like you are using a newer nixpkgs than me and your pipewire requires a newer version of libcamera than is pinned in this project. This project pins libcamera to the raspberry pi fork so that rpi camera stuff will work as it does on official images. If you don't care about that you can disable it with `raspberry-pi-nix.libcamera-overlay.enable = false` ([option docs](https://github.com/tstat/raspberry-pi-nix/blob/master/rpi/default.nix#L20-L29)).

closed 2024-05-08 by tstat

PowerUser64, 2024-05-11:
Whew, finally got everything compiled. Good news, disabling the libcamera overlay worked! I was able to compile the kernel and restart with no issue! Thank you!

tstat, 2024-05-11:
No problem! I'm glad you got it working.

tstat, 2024-06-16:
Libcamera (along with all other packages provided by this repo, except optionally the kernel) is simply provided in an overlay and does not fix the build inputs. So, it is easy to find versions of nixpkgs for which packages do not build.

adminy, 2024-06-16:
So what's the best course of action? Pinning ***or*** Going with whatever is latest and occasionally fix what becomes broken?