nix-community/raspberry-pi-nix #16: rpi5 gets stuck on uboot logo

ghostbuster91, 2024-06-02:
Hi, first of all, thanks for your work on this project :pray: 

I've just tried running nixos from the example repository (both build from sources and using the published images) but my rpi5 gets stuck on the first screen with uboot logo. I am not sure how to debug it since there are no logs or anything like that. Any idea what could be the reason? The default rpi system is running fine. Not sure if that matters but I have a nvme hat installed (with the rpi system on it).

tstat, 2024-06-03:
That's interesting -- I'm not sure why that would be, but if I were you I would just boot with the normal rpi boot process by disabling uboot `raspberry-pi-nix.uboot.enable = false` ([option doc](https://github.com/tstat/raspberry-pi-nix/blob/master/rpi/default.nix#L41-L56)). The main disadvantage is that you can't switch generations in the bootloader, that's kindof difficult with uboot anyway though.

@rszyma I have always seen the uboot generation selection, but I previously experienced a different issue where my usb keyboard presses weren't recognized by uboot, so I could only interact with uboot/pick a different generation over uart.

jDmacD, 2024-06-15:
same happening to me. I have two identical rpi5s. The u-boot logo will show in the corner,  with no generations menu. It feels like the longer I leave it, the more likely it is to boot successfully after a reset. Is there anyway of debugging this with a serial connection or something?

adminy, 2024-06-16:
same here, Sounds like uboot doesn't work with pi 5 properly yet. I see logo, never got past the logo screen. trying without uboot enabled now.

tstat, 2024-06-17:
>  Is there anyway of debugging this with a serial connection or something?

Yes, uboot writes to uart. I have used that on rpi4 a good bit, but not yet on rpi5.