nix-community/raspberry-pi-nix #18: 24.05 don't have any wireless interface on RPI4.

gungun974, 2024-06-10:
Note, all of this is made with a RPI4

Nice project. I recently got one of my nix raspberry pi goes insane with bluetooth but with your flake I successfully make bluetooth re work.

However I downgread my version of Nixpkgs to 23.11 since in unstable and 24.05 wlan0 never appear and a time out of 1minutes and 30 stop the boot.

I tough I could counter this issue temporary with an official raspberry pi dongle but it's seem there is an issue with the wireless module.

I can't really help a lot here for more detail information since I don't have the time to discuss but an easy way to reproduce is taking the starting flake and go to 24.05 branch and just boot.

If nix is truly reproducible you should get the same weird no wlan0 and etc...

Have a nice day ^^

tshakah, 2024-06-11:
The wlan interface is also missing on the pi 5. Happy to run any commands to help debug

tshakah, 2024-06-11:
dmesg shows `brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50`

MinerSebas, 2024-06-16:
I think there is a issue with loading the firmware. I tried two other USB-Wifi-Adpaters i have and both complained about failing to load Firmware.
```
[92630.515344] usb 3-2: new high-speed USB device number 3 using xhci-hcd
[92630.670968] usb 3-2: New USB device found, idVendor=0bda, idProduct=b720, bcdDevice= 2.00
[92630.679190] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[92630.686359] usb 3-2: Product: 802.11n WLAN Adapter
[92630.691171] usb 3-2: Manufacturer: Realtek
[92630.695282] usb 3-2: SerialNumber: 00e04c000001
[92630.705881] usb 3-2: This Realtek USB WiFi dongle (0x0bda:0xb720) is untested!
[92630.713194] usb 3-2: Please report results to Jes.Sorensen@gmail.com
[92630.720971] Bluetooth: hci0: RTL: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[92630.730969] Bluetooth: hci0: RTL: rom_version status=0 version=1
[92630.737041] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723b_fw.bin
[92630.743156] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_fw.bin failed with error -2
[92630.752032] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8723b_fw not found
[92631.201619] usb 3-2: Dumping efuse for RTL8723BU (0x200 bytes):
[92631.207576] 00000000: 29 81 03 7c 01 08 21 00 40 07 05 35 10 00 00 00  )..|..!.@..5....
[92631.215618] 00000010: 2d 2d 2d 2c 2c 2c 2e 2e 2e 2d 2d f3 ff ff ff ff  ---,,,...--.....
[92631.223661] 00000020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.231703] 00000030: ff ff ff ff ff ff ff ff ff ff 2d 2d 2d 2d 2d 2d  ..........------
[92631.239745] 00000040: 2d 2d 2d 2d 2d 00 ff ff ff ff ff ff ff ff ff ff  -----...........
[92631.247786] 00000050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.255826] 00000060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.263867] 00000070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.271907] 00000080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.279948] 00000090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.287989] 000000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.296030] 000000b0: ff ff ff ff ff ff ff ff 21 1f 21 00 00 00 ff ff  ........!.!.....
[92631.304071] 000000c0: ff 28 20 11 00 00 00 ff 00 ff ff ff ff ff ff ff  .( .............
[92631.312112] 000000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.320152] 000000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.328193] 000000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.336234] 00000100: da 0b 20 b7 e7 47 03 74 da 38 5f 74 84 09 03 52  .. ..G.t.8_t...R
[92631.344275] 00000110: 65 61 6c 74 65 6b 16 03 38 30 32 2e 31 31 6e 20  ealtek..802.11n 
[92631.352315] 00000120: 57 4c 41 4e 20 41 64 61 70 74 65 72 00 ff ff ff  WLAN Adapter....
[92631.360356] 00000130: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.368399] 00000140: ff ff ff ff ff ff ff 0f ff ff ff ff ff ff ff ff  ................
[92631.376440] 00000150: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.384481] 00000160: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.392522] 00000170: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.400563] 00000180: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.408604] 00000190: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.416645] 000001a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.424686] 000001b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.432727] 000001c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.440767] 000001d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.448808] 000001e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.456849] 000001f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[92631.464890] usb 3-2: RTL8723BU rev E (SMIC) romver 0, 1T1R, TX queues 3, WiFi=1, BT=1, GPS=0, HI PA=0
[92631.474155] usb 3-2: RTL8723BU MAC: 74:da:38:5f:74:84
[92631.479228] usb 3-2: rtl8xxxu: Loading firmware rtlwifi/rtl8723bu_nic.bin
[92631.486070] usb 3-2: Direct firmware load for rtlwifi/rtl8723bu_nic.bin failed with error -2
[92631.494552] usb 3-2: request_firmware(rtlwifi/rtl8723bu_nic.bin) failed
[92631.501196] usb 3-2: Fatal - failed to load firmware
[92631.506191] rtl8xxxu: probe of 3-2:1.2 failed with error -11
```

Blindly guessing but perhaps this is a clue
https://github.com/tstat/raspberry-pi-nix/blob/26d6ba2a91e6384252bfea5e94c3364e7f8879f2/overlays/default.nix#L37-L39
This project disables xz compression, but with [#nixos/nixpkgs/302300](https://github.com/NixOS/nixpkgs/pull/302300), kernels 5.19+ use zstd for compression instead.

MinerSebas, 2024-06-16:
My hunch was correct.
Adding this to the configuration
```
nixpkgs.overlays = [(_:_: { compressFirmwareZstd = x: x; })];
```
makes the wifi interfacees show up again.

The problem is that old/pinned kernel doesn't have the ability to load zstd compressed firmware.

adminy, 2024-06-16:
but how come everything works fine on the pi 5 then? I haven't disabled compression.

MinerSebas, 2024-06-16:
> but how come everything works fine on the pi 5 then? I haven't disabled compression.

On my pi 5 that fix was necessary for getting wifi to work again.
I have no idea why this seems to work for you anyway.

tshakah, 2024-06-16:
My pi 5 didn't work either, I just built an image with your fix and it found the WiFi. Thanks!

adminy, 2024-06-16:
this is what I've got:
```nix
{
  description = "Flake For SD images";
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/9a9960b98418f8c385f52de3b09a63f9c561427a";
    pi = { url = "github:tstat/raspberry-pi-nix"; inputs.nixpkgs.follows = "nixpkgs"; };
  };
  outputs = inputs@{ self, nixpkgs, pi, ... }: let 
    system = "aarch64-linux";
    pi-config = { pkgs, lib, ... }: {
      system.stateVersion = "24.11";
      time.timeZone = "Ireland/Dublin";
      networking.wireless = {
	enable = true;
	  networks.SSID_NAME.psk = "mypassword";
      };
      services.openssh.enable = true;
      raspberry-pi-nix = {
        libcamera-overlay.enable = false;
        uboot.enable = false;
        pin-kernel.enable = false;
      };
      hardware = {
        bluetooth.enable = true;
        raspberry-pi.config = {
          all = {
            options = {
              arm_64bit = { enable = true; value = true; };
              enable_uart = { enable = true; value = true; };
              avoid_warnings = { enable = true; value = true; };
              camera_auto_detect = { enable = true; value = true; };
              display_auto_detect = { enable = true; value = true; };
              disable_overscan = { enable = true; value = true; };
            };
            dt-overlays = {
              # vc4-kms-v3d = { enable = true; params = {}; };
              vc4-kms-v3d-pi5 = { enable = true; params = {}; };
            };
            base-dt-params = {
              krnbt = { enable = true; value = "on"; };
              spi = { enable = true; value = "on"; };
              # pciex1 = { enable = true; }; # Broken due to missing value param
              pciex1_gen = { enable = true; value = 3; };
            };
          };
        };
      };
    };
    hosts = {
      box_8 = {
        modules = [ pi.nixosModules.raspberry-pi pi-config ];
        hardware = "pi5";
        sshPort = 2223;
        tags = [ "pi 5", "matter" ];
      };
    };
    lib = nixpkgs.lib;
    pkgs = import nixpkgs { inherit system; };
  in {
    # Build nix systems
    nixosConfigurations = lib.mapAttrs' (name: config: 
      lib.nameValuePair "${name}-${config.hardware}" (lib.nixosSystem {
        inherit system;
        modules = config.modules ++ [ ({ ... }: {
          networking = {
            hostName = name;
            firewall.allowedTCPPorts = [ config.sshPort ];
          };
          services.openssh.ports = [ config.sshPort ];
        }) ];
      })) hosts;

    # create image files
    packages = {
      me = self.nixosConfigurations.box_8-pi5.config.system.build.sdImage;
    };
  };
}
```

MinerSebas, 2024-06-16:
I suspect this is due to your `pin-kernel.enable = false;`, which allows you to have a newer kernel with zstd support.
I cannot confirm this, as i am also hit by #19 .

adminy, 2024-06-16:
I'm not sure about that, setting it pinned or not makes no difference to my build for some reason, possibly because nixpkgs is already fixed version.

I'm also hit by the gpu patch, but only if I point to a newer nixpkgs. So the gpu patch implies that upstream nixpkgs is broken in master :o ?

PR #22 (2024-06-16): Also dont use zstd for firmware compression
Fixes #18
closed 2024-06-16 by tstat