nix-community/raspberry-pi-nix #19: [24.05] Build

SebTM, 2024-06-12:
Hey, first of all thanks for the work already done 🙏🏻

I'm currently getting the following issue - I'm trying to understand:
```
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/mfxpz68z7s2vdmxmicn7cgskir3r0nym-linux-config-6.6.28.drv^*'
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/hdh0igwg2rj6zk6xmf8kdn7kzrv9i91h-source
source root is source
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
applying patch /nix/store/mnlqwys9dxznjcvq4w3s0435s9ql8qv8-bridge-stp-helper.patch
patching file net/bridge/br_private.h
Hunk #1 succeeded at 51 with fuzz 2 (offset 12 lines).
applying patch /nix/store/mbz5gln1q4swlzmqs2399br98qjif9kn-request-key-helper.patch
patching file security/keys/request_key.c
Hunk #1 succeeded at 117 (offset 3 lines).
applying patch /nix/store/831w9jjwfspn3wc2h3mlks980v3zm2qz-6de0bb51929cd3ad4fa27b2a421a2af12e6468f5.patch
patching file drivers/gpu/drm/rp1/rp1-dpi/Kconfig
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-dpi/Kconfig.rej
patching file drivers/gpu/drm/rp1/rp1-dsi/Kconfig
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-dsi/Kconfig.rej
patching file drivers/gpu/drm/rp1/rp1-vec/Kconfig
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-vec/Kconfig.rej
```

This happens when switching from 23.11 to 24.05, I'm not sure what's the issue here as the repo does not apply custom-patches?

General Question: Would you be interesting to make this project part of https://github.com/nix-community/?

What are your thoughts on any kind of GitHub-CI and automated builds/push to a binary-cache?

MinerSebas, 2024-06-16:
I have/had the same Problem.
I was trying to build the kernel while using the unstable nixpkgs on RaspberryPi os, and i got the same error. Then i tired it with 23.11 instead which succeded.

The strange thing is after running Nixos 23.11 on my Pi 5, i could switch without this Problem to unstable. (Though i encountered #18 instead)

Then i tried building again the kernel on the unstable Nixos:

```
$ nix build .\#nixosConfigurations.miraculix.pkgs.rpi-kernels.latest.kernel --rebuild
warning: Git tree '/home/sebastian/nix-config' is dirty
these 3 derivations will be built:
  /nix/store/qgcs6bqq1a3kydm6z71rgn53zzwnc6mv-linux-config-6.6.28.drv
  /nix/store/vs6vc0i0cg5ryh1jxjgl1qsnnjbz4y66-installkernel.drv
  /nix/store/whrm83nl3jdn56fgkphid8ksds39vph0-linux-rpi-6.6.28.drv
these 27 paths will be fetched (10.83 MiB download, 53.70 MiB unpacked):
  /nix/store/qfk17ly5bxdw2xmn07jagwwc1544bdib-693a5e69bddbcbe1d1b796ebc7581c3597685b1b.patch
  /nix/store/831w9jjwfspn3wc2h3mlks980v3zm2qz-6de0bb51929cd3ad4fa27b2a421a2af12e6468f5.patch
  /nix/store/4psbpl9n6wg3grxby7drdpk47f2rzpnp-bc-1.07.1
  /nix/store/x7ihrrp0cpnldm0jgz70dxr09w4spzli-bison-3.8.2
  /nix/store/cmyp0aa3hw005vi2cawf7y9p82b4f6ab-elfutils-0.191-bin
  /nix/store/3904i4m12wn0iy3mrvxh07jhhlp4xwyv-elfutils-0.191-dev
  /nix/store/fdfc9dkg8njgvgpv8a39hawyicxc8kpa-file-5.45
  /nix/store/qs0k2m9lx9zw7i0p7yfpp3fhd4yhk5qm-flex-2.6.4
  /nix/store/45vmmh96gqz026ijnf6v3wj2z0yf66hl-gmp-with-cxx-6.3.0-dev
  /nix/store/zrv00wxxwq6l1kwdlhwbkby4xxbfw3i9-gnu-config-2024-01-01
  /nix/store/6npgxawv4j2nbg5511m1pyms6kz9nzig-gnum4-1.4.19
  /nix/store/3w7ldfr5lrwmvfkhgmvgiq0dsb15f87a-gnumake-4.4.1
  /nix/store/bjkvp6bcj2gk7rwq3fvmps5llz5y7c9s-kmod-31-dev
  /nix/store/hk8hx943k3rc2jh0jiyxspqhi8n0z6nd-libffi-3.4.6
  /nix/store/jd922bs0bxr805r9ndkx7y0jl0n2wwc0-libmpc-1.3.1
  /nix/store/gqipvcpld5sdj71cslb1nglwdspyw0lj-mpfr-4.2.1
  /nix/store/6yhl7qaqp1ka7hnxdznd6bjgjjhc05ws-mpfr-4.2.1-dev
  /nix/store/i0mly410wz8sn1xmvyh0dhk28hz0bvpd-openssl-3.0.13-bin
  /nix/store/znh18bsvgl91k16z7rsjpm9zz3kyqxfp-openssl-3.0.13-dev
  /nix/store/31643mr0css6ll9a36a13gnqb0gqch5g-pahole-1.26
  /nix/store/7imcddzn9w5zq9q9kbc5h5riwwqvxnv1-python3-minimal-3.11.9
  /nix/store/67fwh8xzp3bs4mll94x6l4h9kr8mfim5-setup-debug-info-dirs-hook
  /nix/store/hyaqjk5jdavx2zjk3s79ahz87xxcz0a2-stdenv-linux
  /nix/store/mia67025vf08askm8jlz4l08k45rgapn-stdenv-linux
  /nix/store/nfz2sh9nsfi0g0iw14vi1jzcarqk20vd-update-autotools-gnu-config-scripts-hook
  /nix/store/4f32crjfmzlzknnaw52hdjqix94gd605-zlib-1.3.1-dev
  /nix/store/66b36wbm0rialh0fn547dgvkndyw6vix-zstd-1.5.6-dev
error: some outputs of '/nix/store/whrm83nl3jdn56fgkphid8ksds39vph0-linux-rpi-6.6.28.drv' are not valid, so checking is not possible
```

The kernel package is broken ?!?
Ok, try it again with the --repair flag.

```
$ nix build .\#nixosConfigurations.miraculix.pkgs.rpi-kernels.latest.kernel --rebuild --repair
warning: Git tree '/home/sebastian/nix-config' is dirty
these 3 derivations will be built:
  /nix/store/qgcs6bqq1a3kydm6z71rgn53zzwnc6mv-linux-config-6.6.28.drv
  /nix/store/vs6vc0i0cg5ryh1jxjgl1qsnnjbz4y66-installkernel.drv
  /nix/store/whrm83nl3jdn56fgkphid8ksds39vph0-linux-rpi-6.6.28.drv
these 27 paths will be fetched (10.83 MiB download, 53.70 MiB unpacked):
  /nix/store/qfk17ly5bxdw2xmn07jagwwc1544bdib-693a5e69bddbcbe1d1b796ebc7581c3597685b1b.patch
  /nix/store/831w9jjwfspn3wc2h3mlks980v3zm2qz-6de0bb51929cd3ad4fa27b2a421a2af12e6468f5.patch
  /nix/store/4psbpl9n6wg3grxby7drdpk47f2rzpnp-bc-1.07.1
  /nix/store/x7ihrrp0cpnldm0jgz70dxr09w4spzli-bison-3.8.2
  /nix/store/cmyp0aa3hw005vi2cawf7y9p82b4f6ab-elfutils-0.191-bin
  /nix/store/3904i4m12wn0iy3mrvxh07jhhlp4xwyv-elfutils-0.191-dev
  /nix/store/fdfc9dkg8njgvgpv8a39hawyicxc8kpa-file-5.45
  /nix/store/qs0k2m9lx9zw7i0p7yfpp3fhd4yhk5qm-flex-2.6.4
  /nix/store/45vmmh96gqz026ijnf6v3wj2z0yf66hl-gmp-with-cxx-6.3.0-dev
  /nix/store/zrv00wxxwq6l1kwdlhwbkby4xxbfw3i9-gnu-config-2024-01-01
  /nix/store/6npgxawv4j2nbg5511m1pyms6kz9nzig-gnum4-1.4.19
  /nix/store/3w7ldfr5lrwmvfkhgmvgiq0dsb15f87a-gnumake-4.4.1
  /nix/store/bjkvp6bcj2gk7rwq3fvmps5llz5y7c9s-kmod-31-dev
  /nix/store/hk8hx943k3rc2jh0jiyxspqhi8n0z6nd-libffi-3.4.6
[... 68 lines cut ...]
linux-config> Skipping patch.
linux-config> 1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-dsi/Kconfig.rej
linux-config> patching file drivers/gpu/drm/rp1/rp1-vec/Kconfig
linux-config> Reversed (or previously applied) patch detected!  Assume -R? [n] 
linux-config> Apply anyway? [n] 
linux-config> Skipping patch.
linux-config> 1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-vec/Kconfig.rej
error: builder for '/nix/store/qgcs6bqq1a3kydm6z71rgn53zzwnc6mv-linux-config-6.6.28.drv' failed with exit code 1;
       last 10 log lines:
       > patching file drivers/gpu/drm/rp1/rp1-dsi/Kconfig
       > Reversed (or previously applied) patch detected!  Assume -R? [n]
       > Apply anyway? [n]
       > Skipping patch.
       > 1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-dsi/Kconfig.rej
       > patching file drivers/gpu/drm/rp1/rp1-vec/Kconfig
       > Reversed (or previously applied) patch detected!  Assume -R? [n]
       > Apply anyway? [n]
       > Skipping patch.
       > 1 out of 1 hunk ignored -- saving rejects to file drivers/gpu/drm/rp1/rp1-vec/Kconfig.rej
       For full logs, run 'nix log /nix/store/qgcs6bqq1a3kydm6z71rgn53zzwnc6mv-linux-config-6.6.28.drv'.
error: 1 dependencies of derivation '/nix/store/whrm83nl3jdn56fgkphid8ksds39vph0-linux-rpi-6.6.28.drv' failed to build
```
The same Error...

Honestly i don't know if it helps, but something strange is going on here.

tstat, 2024-06-16:
This could arise if you are using a different kernel than is pinned in this project. If you are using the latest release of this project (`v0.3.1`) and have the default setting of `raspberry-pi-nix.pin-kernel = true` then this shouldn't be possible.

Could you share what commit of `raspberry-pi-nix` you are using?

tstat, 2024-06-16:
@SebTM 

> Would you be interesting to make this project part of https://github.com/nix-community/?

I never thought about that, but it sounds good to me.

>  What are your thoughts on any kind of GitHub-CI and automated builds/push to a binary-cache?

That would be a great help

SebTM, 2024-06-22:
Hey, sorry for the delay on my side 🙏🏻 

> Could you share what commit of raspberry-pi-nix you are using?
I was using latest master at the time so I guess `26d6ba2a91e6384252bfea5e94c3364e7f8879f2`

> I never thought about that, but it sounds good to me.
I've asked in the matrix-channel what's the process to move forwards as I really like the idea to centralize efforts on RPI-Nix for the community 😃 

(https://matrix.to/#/!PbtOpdWBSRFbEZRLIf:numtide.com/$g5UXWWioNoxTvbrD1vq87ko2Quv1JnhG8p8mBiOfNww?via=blad.is&via=matrix.org&via=nixos.dev)

> That would be a great help
I will have a look, do you think cachix free-tier (5 GB) is enough otherwise I can offer a self-hosted binary-cache to avoid costs ✌🏻 

---

Building against 24.05 with my config I used when opening the above issue and the example I tried to reproduce works fine now (against master, `tstat/raspberry-pi-nix/6045de5f79ac8eb432cd75e82e1accfb48caff45`) so I would leave it open for the other topics until finished.

tstat, 2024-06-22:
> I will have a look, do you think cachix free-tier (5 GB) is enough otherwise I can offer a self-hosted binary-cache to avoid costs ✌🏻

Thanks for the offer, but I don't think that's necessary. I do push kernel builds to the cachix free tier, and I suggest that others may use it to avoid compiling the kernel [in the readme](https://github.com/tstat/raspberry-pi-nix/blob/master/README.md?plain=1#L83-L102). It is also used [in the example repo](https://github.com/tstat/raspberry-pi-nix-example/blob/master/flake.nix#L11).

I don't have any sort of CI to do the build though, as mentioned above that is just me. I'm not aware of any free services that I could use.

tstat, 2024-06-22:
As mentioned above, the issues described here would occur on `v0.3.0`, but were fixed with `v0.3.1` when the references to the unpinned kernel were removed. So, I suspect you all had a dependency on `v0.3.0` when experiencing this issue. If you have a repro with a newer version feel free to reopen.

closed 2024-06-22 by tstat

zowoq, 2024-06-23:
> I've asked in the matrix-channel what's the process to move forwards as I really like the idea to centralize efforts on RPI-Nix for the community

@tstat I've sent you an invite to join @nix-community and created a team @nix-community/raspberry-pi-nix.

We have a aarch64-linux CI builder and 1tb cachix cache if you want to use it.

https://github.com/nix-community/infra/blob/master/devdoc/hosts.md#build04
https://nix-community.org/continuous-integration/

I'm not really familiar with all the raspberry pi / SBC stuff, the nixos community builder has a kernel patch for arm7l builds, if that is useful I could look at applying that to our CI builder.

https://github.com/NixOS/aarch64-build-box/blob/master/armv7l.nix

tstat, 2024-06-23:
Oh, that sounds great! Thanks @zowoq

tstat, 2024-06-24:
> I'm not really familiar with all the raspberry pi / SBC stuff, the nixos community builder has a kernel patch for arm7l builds, if that is useful I could look at applying that to our CI builder.

No need on my account. This repo only attempts to support aarch64, which covers all recent (past ~6 years or so) rpi sbcs.

PR #23 (2024-07-08): chore: bump libs & fix libcamera, fix upstream nixpkgs
things tested and working:
- [x] boots on pi 5 b
- [x] libcamera works
- [x] nixpkgs master works

things to try out:
 - ~[ ] uboot booting, see if its fixed.~ Uboot still not working
      - [ ] test sd card boot
 
[image]

 - [ ] pi 4 test.

This should fix #19 
Untested but maybe also #16 

### Serial error
```bash
DRAM:  1016 MiB (effective 8 GiB)
RPI 5 Model B (0xd04170)
Core:  22 devices, 11 uclasses, devicetree: board
MMC:   mmc@fff000: 0, mmc@1100000: 1
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:  [... cut]