nix-community/raspberry-pi-nix #27: Have an option to disable/enable specific features (ex: libcamera)

koalalorenzo, 2024-07-11:
Is there any way to disable rpicam-apps and libcamera?

I have an issue that when I am building the sd image for my RPis (4 and 5) but for some reason the libcamera build fails, with this error:

```log
[183/209] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcamerapad.cpp.o
[184/209] Generating src/py/libcamera/py_gen_controls with a custom command
FAILED: src/py/libcamera/py_controls_generated.cpp
/build/source/src/py/libcamera/gen-py-controls.py --mode controls -o src/py/libcamera/py_controls_generated.cpp -t ../src/py/libcamera/py_controls_generated.>
/bin/sh: /build/source/src/py/libcamera/gen-py-controls.py: not found
[185/209] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcamerapool.cpp.o
[186/209] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcameraprovider.cpp.o
[187/209] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcamerasrc.cpp.o
ninja: build stopped: subcommand failed.
```

Since I don't need the camera (as those are part of a Nomad cluster) can I just disable these modules somehow?

---

This is my flake:

```nix
{
  description = "nixos generator and raspberrypis image builders";
  
  nixConfig = {
    extra-substituters = [ "https://raspberry-pi-nix.cachix.org" ];
    extra-trusted-public-keys = [
      "raspberry-pi-nix.cachix.org-1:WmV2rdSangxW0rZjY/tBvBDSaNFQ3DyEQsVw8EvHn9o="
    ];
  };
  
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.05";
    nixos-hardware.url = "https://github.com/NixOS/nixos-hardware/archive/master.tar.gz";
    raspberry-pi-nix = {
      url = "github:tstat/raspberry-pi-nix";
    };
    sops-nix.url = "github:Mic92/sops-nix";
    home-manager.url = "https://github.com/nix-community/home-manager/archive/release-24.05.tar.gz";
    nixpkgs-unstable.url = "github:NixOS/nixpkgs/nixpkgs-unstable";
    nixos-generators = {
      url = "github:nix-community/nixos-generators";
      inputs.nixpkgs.follows = "nixpkgs";
    };
  };
  outputs = { self, nixpkgs, nixos-hardware, raspberry-pi-nix, sops-nix, home-manager, nixpkgs-unstable, nixos-generators, ... }:
    let
      inherit (nixpkgs.lib) nixosSystem;
      base-setup = { pkgs, lib, ... }: {
        system.stateVersion = "24.05";
        networking.hostName = "nixos";
        nixpkgs.config = {
          packageOverrides = pkgs: {
            unstable = import <nixos-unstable> {
              config = pkgs.config;
            };
          };
        };

      };
      rpi-config = { pkgs, lib, ... }: {
        fileSystems."/".options = [ "noatime" ];
        boot.initrd.availableKernelModules = [ "xhci_pci" "usbhid" "usb_storage" ];

        # Wait for uptime so that it syncs properly
        systemd.additionalUpstreamSystemUnits = [ "systemd-time-wait-sync.service" ];
        systemd.services.systemd-time-wait-sync.wantedBy = [ "multi-user.target" ];

        console.enable = false;
        environment.systemPackages = with pkgs; [
          libraspberrypi
          raspberrypi-eeprom
        ];

        
        hardware = {
          raspberry-pi = {
            config = {
              all = {
                options.camera_auto_detect = {
                  enable = false;
                };
                
                base-dt-params = {
                  # enable autoprobing of bluetooth driver
                  # https://github.com/raspberrypi/linux/blob/c8c99191e1419062ac8b668956d19e788865912a/arch/arm/boot/dts/overlays/README#L222-L224
                  krnbt = {
                    enable = true;
                    value = "on";
                  };
                };
              };
            };
          };
        };
      };

    in {
      nixosConfigurations = {
        rpi4 = nixosSystem {
          system = "aarch64-linux";
          modules = [
            sops-nix.nixosModules.sops
            base-setup
            raspberry-pi-nix.nixosModules.raspberry-pi
            rpi-config
            ./rpi4.nix
            ./networking.nix
            ./common.nix
          ];
        };
        rpi5 = nixosSystem {
          system = "aarch64-linux";
          modules = [ 
            sops-nix.nixosModules.sops
            base-setup
            raspberry-pi-nix.nixosModules.raspberry-pi
            rpi-config
            ./rpi5.nix
            ./networking.nix
            ./common.nix
          ];
        };
      };
    };
}

```

koalalorenzo, 2024-07-15:
I was not using a specific version, I did not specify any, I just edited from the example in the README

tstat, 2024-07-24:
I believe my previous comment highlights the option you want to set. Let me know if you are having trouble.

closed 2024-07-24 by tstat