nix-community/raspberry-pi-nix #30: Module `tpm-crb` not found in directory

Yash-Garg, 2024-07-24:
I get this error while trying to build an `sdImage`, setting `boot.initrd.systemd.enableTpm2 = lib.mkForce false;` works but just confirming if this is expected? For some reason the built `sdImage` doesn't boot for me on Raspberry Pi 5. 

[My Config](https://github.com/Yash-Garg/dotfiles/tree/feat/raspi5)

```nix
trace: warning: lib.cartesianProductOfSets is a deprecated alias of lib.cartesianProduct.
error: builder for '/nix/store/4g3kyfzkibw014f4639wl1q8yvqvf60s-linux-rpi-6.6.31-modules-shrunk.drv' failed with exit code 1;
       last 25 log lines:
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-timer.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm-dmaengine.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/ac97_bus.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-compress.ko.xz
       >   dependency already copied: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/snd-soc-core.ko.xz
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/codecs/snd-soc-hdmi-codec.ko.xz
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/vc4/vc4.ko.xz
       > root module: pcie_brcmstb
       >   builtin dependency: pcie_brcmstb
       > root module: reset-raspberrypi
       >   builtin dependency: reset_raspberrypi
       > root module: ext2
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/fs/ext2/ext2.ko.xz
       > root module: ext4
       >   builtin dependency: ext4
       > root module: autofs
       >   builtin dependency: autofs4
       > root module: tpm-tis
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/char/tpm/tpm.ko.xz
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/char/tpm/tpm_tis_core.ko.xz
       >   copying dependency: /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/char/tpm/tpm_tis.ko.xz
       > root module: tpm-crb
       > modprobe: FATAL: Module tpm-crb not found in directory /nix/store/5640ym8zpbckqklgrr8hpk21ik9yp2mp-linux-rpi-6.6.31-modules/lib/modules/6.6.31
       For full logs, run 'nix log /nix/store/4g3kyfzkibw014f4639wl1q8yvqvf60s-linux-rpi-6.6.31-modules-shrunk.drv'.
error: 1 dependencies of derivation '/nix/store/cnknyxi80lxpp1mm80awz7lg5w72k8j0-initrd-linux-rpi-6.6.31.drv' failed to build
error: 1 dependencies of derivation '/nix/store/zn5mcx7ps1nqj6f3j2kyhzr9vifw7h6k-nixos-system-eclipse-24.11.20240722.453402b.drv' failed to build
error: 1 dependencies of derivation '/nix/store/0svlb0v2x3g5mccab3adp5rkj5q1q2aj-ext4-fs.img.zst.drv' failed to build
error: 1 dependencies of derivation '/nix/store/kn34mi10sdkb5swqjfmiy1qvqrhfj31h-nixos-sd-image-24.11.20240722.453402b-aarch64-linux.img.drv' failed to build
```

closed 2024-07-24 by Yash-Garg

bddvlpr, 2024-07-30:
I'm still getting root=fstab is invalid. How did you resolve this?

Yash-Garg, 2024-07-31:
> I'm still getting root=fstab is invalid. How did you resolve this?

In my case it was due to a `mixin` I was using from [srvos](https://nix-community.github.io/srvos/nixos/type/#server-nixosmodulesserver), removing that fixed it for me, [`hardware-configuration.nix`](https://github.com/Yash-Garg/dotfiles/blob/stable/systems/aarch64-linux/cosmos/hardware-configuration.nix) is here for reference.

bddvlpr, 2024-07-31:
I ended up just disabling systemd on initrd like so; `boot.initrd.systemd.enable = false;` so I could keep the srvos module.