nix-community/raspberry-pi-nix #31: Pi 4 bootloops on kernel 6.6.31

tshakah, 2024-07-24:
Came across this today when I updated a pi 4 (pi 5 worked fine). Really stumped me, and then I noticed this issue in nixpkgs: https://github.com/NixOS/nixpkgs/issues/325473.

This is mostly an FYI for anyone else if people look here for issues and don't think to look in nixpkgs too

tstat, 2024-07-24:
That shouldn't be happening on this repo, which a pinned version of rpi's kernel fork. Are you using `raspberry-pi-nix` master and setting the `raspberry-pi-nix.board` option to `bcm2711`? It defaults to `bcm2712` now, which is a mistake I think. We should have no default and require the user to specify I think.

tshakah, 2024-07-24:
Thanks for the quick reply! I was chatting to a colleague about it and we decided to try a different kernel version, and they noticed the board being set to `bcm2712` when looking through this repo's code to see how to do that. They were going to try change it and get back to me, so it's good to know that should work.

tshakah, 2024-07-25:
Changing the board worked, thanks.

closed 2024-07-25 by tshakah