nix-community/raspberry-pi-nix #35: Various `Module not found in directory` errors

K-Finlay, 2024-07-26:
I am getting build errors when setting `raspberry-pi-nix.board = "bcmrpi3`.
The initial one I get is `Module pcie_brcmstb not found in directory`.

The Raspberry Pi 3 / Zero 2 W don't support PCIE, so I removed pcie_brcmstb from `initrd.availableKernelModules`, and that fixed that error.

However, now I am hitting the same issue, but with ahci.

`bcm2712` and `bcm2711` build just fine.
Haven't tried 32-bit builds.

Here are the full error messages:

```
kernel version is 6.6.31
root module: usbhid
  builtin dependency: usbhid
root module: usb_storage
  builtin dependency: usb_storage
root module: vc4
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-timer.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm-dmaengine.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/ac97_bus.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-compress.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/snd-soc-core.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/drm_dma_helper.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/media/cec/core/cec.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/display/drm_display_helper.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-timer.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm-dmaengine.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/ac97_bus.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-compress.ko.xz
  dependency already copied: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/snd-soc-core.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/codecs/snd-soc-hdmi-codec.ko.xz
  copying dependency: /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/vc4/vc4.ko.xz
root module: pcie_brcmstb
modprobe: FATAL: Module pcie_brcmstb not found in directory /nix/store/b73wg805kh2pc94ngzqfkb1m1ff19znj-linux-6.6.31-modules/lib/modules/6.6.31
```

```
kernel version is 6.6.31
root module: usbhid
  builtin dependency: usbhid
root module: vc4
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-timer.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm-dmaengine.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/ac97_bus.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-compress.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/snd-soc-core.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/drm_dma_helper.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/media/cec/core/cec.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/display/drm_display_helper.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-timer.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-pcm-dmaengine.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/ac97_bus.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/core/snd-compress.ko.xz
  dependency already copied: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/snd-soc-core.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/sound/soc/codecs/snd-soc-hdmi-codec.ko.xz
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/drivers/gpu/drm/vc4/vc4.ko.xz
root module: ext2
  copying dependency: /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31/kernel/fs/ext2/ext2.ko.xz
root module: ext4
  builtin dependency: ext4
root module: ahci
modprobe: FATAL: Module ahci not found in directory /nix/store/bf52s8b3bp2l6p1s8ay7caly9ixdq1z5-linux-6.6.31-modules/lib/modules/6.6.31
```

tstat, 2024-07-26:
This isn't too surprising, I have only ever built with `bcm2711` and `bcm2712`. That being said, the 3+ and zero 2 w both work fine with bcm2711, and that is the suggested 64bit config for those boards ([rpi docs](https://www.raspberrypi.com/documentation/computers/linux_kernel.html#native-build-configuration)). I personally run a raspberry pi 3+ and zero 2 w with this project. 

In response to this issue I removed all untested board configs 3375c37c2cda13074ede1f5ed12bcb651d0ffc5a.

tstat, 2024-08-22:
Let me know if you run into other issues.

closed 2024-08-22 by tstat