nix-community/raspberry-pi-nix #36: Configuration does not persist across reboots

SED4906, 2024-07-26:
I'm not sure if this is actually just user error, but the process I attempted was:
1. Build an SD image using an installation of Nix under Raspberry Pi OS on a different SD card
2. Write the image to the target SD card and boot from it
3. Log in as `root` and notice there's nothing in `/etc/nixos/` so I run nixos-generate-config
4. The generated config has GRUB enabled for some reason
5. `nixos-rebuild switch` fails because flakes aren't enabled (???)
6. Enable flakes in `root`'s per-user nix.conf
7. A different error about not finding nixos-config
8. `nixos-rebuild -I nixos-config=/etc/nixos/configuration.nix switch`
9. After rebooting the configuration does not persist
This is on a Raspberry Pi 5

tstat, 2024-07-26:
Hi, this does sound like a user error.

It sounds like you generated a default config in (3) that does not import `raspberry-pi-nix`, then switched. If that's correct, then a failure is expected. 

This project supports and uses the default raspberry pi boot process, so you will need to switch to configurations that handle that appropriately. This project does handle that appropriately, so importing it remains important after the initial sd card image is written.

Some of what you mentioned is related to https://github.com/nix-community/raspberry-pi-nix/discussions/32, so you may find that discussion interesting.

SED4906, 2024-07-26:
how do I import `raspberry-pi-nix` in this case?

SED4906, 2024-07-26:
ah, it looks like it just goes in that line at the top

SED4906, 2024-07-26:
what does a working configuration.nix look like

tstat, 2024-07-26:
The [readme links](https://github.com/nix-community/raspberry-pi-nix/blob/master/README.md?plain=1#L22-L24) the example repo https://github.com/tstat/raspberry-pi-nix-example/tree/master.

SED4906, 2024-08-01:
I haven't tried to use this flake in a while, but an update: after booting the built system, I put an adapted version of the example config in the README in `/etc/nixos/flake.nix` and ran `nixos-rebuild switch`, and it tried to rebuild everything again. Including the kernel. I couldn't get it to not do that after multiple attempts, so I gave up.

mroi, 2024-08-01:
Did you include the `flake.lock` file from this repo? Otherwise the rebuild will build against the then-current version of `nixpkgs`, which will be different from the one this repo is locked to.

closed 2024-08-22 by tstat