nix-community/raspberry-pi-nix #37: libcamera build error, gen-py-controls.py not found

dschaub, 2024-07-26:
Hello! We're actually seeing the same build error as in #27, but opening a new issue since their request was different. We _are_ trying to use libcamera, so we're wondering what could be causing this build error. Here's the failure again:

```
[186/209] Generating src/py/libcamera/py_gen_controls with a custom command
FAILED: src/py/libcamera/py_controls_generated.cpp
/build/source/src/py/libcamera/gen-py-controls.py --mode controls -o src/py/libcamera/py_controls_generated.cpp -t ../src/py/libcamera/py_controls_generated.cpp.in ../src/py/libcamera/../../libcamera/control_ids_draft.yaml ../src/py/libcamera/../../libcamera/control_ids_core.yaml ../src/py/libcamera/../../libcamera/control_ids_rpi.yaml
/bin/sh: /build/source/src/py/libcamera/gen-py-controls.py: not found
```

(Likely coming from [this meson.build](https://github.com/raspberrypi/libcamera/blob/6ddd79b5bdbedc1f61007aed35391f1559f9e29a/src/py/libcamera/meson.build#L34).)

We've tried making sure we're on the same version of nixpkgs, and we've tried on multiple machines. This error is while building against master (specifically b4ad649, but also on the new commits as of this morning). We've tried building v0.3.2 and got a different error, but we were interested in the Pi 5 updates and the fixes to libcamera that were merged a few weeks ago.

Happy to provide any more details that can help! Thanks!

dschaub, 2024-07-26:
Here's the complete build log, if it helps:

```@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/s1j6fzny27m2dnvmbwal3z03bcwlcfi5-source
source root is source
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
patching script interpreter paths in utils/
utils/tuning/rkisp1.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/tuning/raspberrypi_alsc_only.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/tracepoints/analyze-ipa-trace.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/tracepoints/gen-tp-header.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipu3/ipu3-capture.sh: interpreter directive changed from "#!/bin/bash" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/bash"
utils/ipu3/ipu3-process.sh: interpreter directive changed from "#!/bin/bash" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/bash"
utils/update-mojo.sh: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/gen-header.sh: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/update-kernel-headers.sh: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/abi-compat.sh: interpreter directive changed from "#!/bin/bash" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/bash"
utils/gen-controls.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/generate.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/parser.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/extract-docs.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/mojo/public/tools/bindings/mojom_bindings_generator.py: interpreter directive changed from "#!/usr/bin/env python" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python"
utils/ipc/mojo/public/tools/bindings/generate_type_mappings.py: interpreter directive changed from "#!/usr/bin/env python" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python"
utils/ipc/mojo/public/tools/bindings/validate_typemap_config.py: interpreter directive changed from "#!/usr/bin/env python" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python"
utils/ipc/mojo/public/tools/bindings/minify_with_terser.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/mojo/public/tools/bindings/concatenate_and_replace_closure_exports.py: interpreter directive changed from "#!/usr/bin/env python" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python"
utils/ipc/mojo/public/tools/bindings/concatenate-files.py: interpreter directive changed from "#!/usr/bin/env python" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python"
utils/ipc/mojo/public/tools/run_all_python_unittests.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/mojo/public/tools/mojom/check_stable_mojom_compatibility.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/mojo/public/tools/mojom/check_stable_mojom_compatibility_unittest.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/ipc/mojo/public/tools/mojom/mojom_parser.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/gen-formats.py: interpreter directive changed from "#!/usr/bin/env python3" to "/nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/bin/python3"
utils/hooks/pre-push: interpreter directive changed from "#!/bin/bash" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/bash"
utils/hooks/post-commit: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/hooks/pre-commit: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/release.sh: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
utils/gen-ipa-priv-key.sh: interpreter directive changed from "#!/bin/sh" to "/nix/store/1fzg4cl3k2n9yq80vg6y1vcmvx3qm682-bash-5.2p26/bin/sh"
[... cut ...]
/build/source/src/py/libcamera/gen-py-controls.py --mode controls -o src/py/libcamera/py_controls_generated.cpp -t ../src/py/libcamera/py_controls_generated.cpp.in ../src/py/libcamera/../../libcamera/control_ids_draft.yaml ../src/py/libcamera/../../libcamera/control_ids_core.yaml ../src/py/libcamera/../../libcamera/control_ids_rpi.yaml
/bin/sh: /build/source/src/py/libcamera/gen-py-controls.py: not found
[187/209] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcamerasrc.cpp.o
ninja: build stopped: subcommand failed.```

dschaub, 2024-07-26:
Update: the error is because gen-py-controls.py can't find python with just the `#!/usr/bin/env python3` shebang, I think. If I modify [this](https://github.com/raspberrypi/libcamera/blob/6ddd79b5bdbedc1f61007aed35391f1559f9e29a/src/py/libcamera/meson.build#L43) line in meson.build to run the script with python3 directly then it works.

I'm fairly new to nix, but I'm still digging around because I know there's a way to [patch shebangs](https://nixos.org/manual/nixpkgs/stable/#patch-shebangs.sh)...

dschaub, 2024-07-26:
Got it! The update that added patching to `src/py` is recent and still in nixpkgs unstable: https://github.com/NixOS/nixpkgs/commit/8a45e936ac04e33133fe118fa1e3eea3cf2a5980. Whew.

SebTM, 2024-07-28:
https://github.com/NixOS/nixpkgs/pull/330393

Opened a backport for the MR updating and fixing the shebang for libcamera on stable - but the overlay (https://github.com/nix-community/raspberry-pi-nix/blob/master/overlays/libcamera.nix#L25) still needs to be updated. Currently (but can be a local issue) with v0.4.0  I don't hit the binary-cache somehow.

tshakah, 2024-07-29:
As a consumer of this project's flake, what would be an idiomatic way of overriding it to use the unstable libcamera, given the MR has been closed?

SebTM, 2024-07-29:
The MR does not really matter as the project has an independent overlay which defines the version used, if @dschaub has it already fixed he can share it via MR here or I will have a look when I find some time :)

tshakah, 2024-08-17:
I believe this can be closed with the merge of https://github.com/nix-community/raspberry-pi-nix/pull/42

closed 2024-08-18 by tstat