nix-community/raspberry-pi-nix #40: Getting CSI camera to work on Raspberry Pi 5?

crabdancing, 2024-07-29:
Has anyone gotten the CSI camera to work on Raspberry Pi 5 with this? I merged [#330393](https://github.com/NixOS/nixpkgs/pull/330393) in my personal copy of nixpkgs stable, and it builds & deploys correctly -- but I can't seem to see the CSI camera no matter how many times I poke around with `v4l2-ctl` or `/dev`. What am I missing?

tshakah, 2024-08-17:
https://github.com/nix-community/raspberry-pi-nix/pull/42 achieves the same thing as the libcamera upgrade, but when a camera is detected running any of the libcamera apps results in a segfault. I haven't gotten any further with it

crabdancing, 2024-08-17:
Hmm. Maybe a dep being behind upstream is causing it?

tyxieblub, 2024-10-20:
I'm also struggling with this: the current flake does not compile since `rpicam-apps` 1.5.2 seems to need `libcamera` 0.3.2 for IMX500 compatibility. But even if downgrading `rpicam-apps` to 1.5.1, when using nixos-24.05, I'm getting a segmentation fault.

I only get an abort with coredump, while complaining about missing `backend_default_config.json` in `libpisp` on nixos-unstable. Correcting does not fix it.

I also tried updating everything, but got nowhere; I will likely try that again soon, but if anyone has any pointers or ideas that would be lovely :D