nix-community/raspberry-pi-nix #43: /dev/root: Can't open blockdev

Svenum, 2024-08-02:
I build the sd-image for the raspberry pi 5 and got this error:

```console
...
Disabling rootwait; root= is invalid
/dev/root: Can't open blockdev
VFS: Cannot open root device "fstab" or unknown-block(0,0): error -6
```

If I look on the sd card I only see the folder "nix" and "sbin" every other folder like "lib", "etc", "tmp", ... is missing.

Any Idea on that?

Everytime I build the image I have to recompile the linux kernel even if I use cachix nix-community cache. So I hope not to rebuild this often.

Here is my config: https://github.com/Svenum/holynix/tree/raspi/systems/aarch64-linux/srv-raspi

SebTM, 2024-08-17:
Hey, you are following your nixpkgs for raspberdy-pi-nix in your flake.nix (https://github.com/Svenum/holynix/blob/raspi/flake.nix#L72C7-L73C1) which makes the cache not hit as we built against the nixpkgs-commit of the lock file at commit time. But I'm not fully certain, if the next issue would be the use of zfs as it's not included in the kernel (and might cause a rebuild) but didn't use it on a pi yet and switched to btrfs in general.

Svenum, 2024-08-18:
Building without zfs works, and not using nixpkgs for raspberry-pi-nix in my flake seems to not compile the kernel, but webkitgtk-2.44.2 is still getting compiled and needs a very long time. Is this also cached?

Svenum, 2024-08-20:
setting `networking.networkmanager.plugins = lib.mkForce []` removed the webkitgtk dependency

closed 2024-08-20 by Svenum