nix-community/raspberry-pi-nix #48: Rebuild error with latest raspberry-pi-nix and nixpkgs 24.05

PowerUser64, 2024-08-18:
I was updating my system, when I got the following error:

```
       error: The option `boot.loader.generic-extlinux-compatible.enable' has conflicting definition values:
       - In `/nix/store/4f9cx80vcs558qa1i68aa74yhc0lxsk5-source/raspberry-pi/4': true
       - In `/nix/store/7sprarsdfz9qcd7859phvr9nvhi14mri-source/flake.nix': false
       Use `lib.mkForce value` or `lib.mkDefault value` to change the priority on any of these definitions.
```

The first file listed in the error is from raspberry-pi-nix, and the second one is from nixpkgs. It seems that raspberry-pi-nix tries to set a value that nixpkgs also sets. I solved this by using `lib.mkForce` in my config, but it seems like something that should be solved in rpi nix or nixpkgs. I only started getting this error recently. Any idea where this should be solved?

PowerUser64, 2024-08-19:
I manually set `boot.loader.generic-extlinux-compatible.enable` to false with lib.mkForce to avoid the second error, and now my pi won't boot :zany_face:

I'm recovering it, with the main disk hooked up to my desktop. I'm not 100% sure what to do to satisfy nix here, but I'll post updates here as I go.

Still battling the same errors as mentioned before. Resolving the first one by setting `boot.loader.generic-extlinux-compatible.enable` (with mkForce) to false leads to no boot, and setting it to true causes the second error above. I can't set `system.build.installBootLoader` to anything because it causes an error saying the option doesn't exist.

PowerUser64, 2024-08-19:
Turns out [rpi-nix looks at the uboot option](https://github.com/nix-community/raspberry-pi-nix/blob/a993f0cd595ecc0836ca82640657527b934f9c9e/rpi/default.nix#L327-L328), and sets `boot.loader.generic-extlinux-compatible.enable` as well as `boot.loader.initScript.enable` automatically. [nixpkgs looks at `boot.loader.initScript.enable`](https://github.com/NixOS/nixpkgs/blob/5d151d4d94415f9f75b4e039b6707af46e5ba6dc/nixos/modules/system/boot/loader/init-script/init-script.nix#L46-L50) as its way of knowing whether to set this mysterious `system.build.installBootLoader` option.

I have uboot disabled (the default), which should cause rpi nix to disable both `initScript` and `generic-extlinux-compatible`. This causes nixpkgs to disable `system.build.installBootLoader` ... I'm going crazy trying to figure this out lol.

Anyway, what's happening is rpi-nix wants to enable an option ([`generic-extlinux-compatible`](https://github.com/NixOS/nixpkgs/blob/f84991dc2bfc98c32174dbc92290d84438342867/nixos/modules/system/boot/loader/generic-extlinux-compatible/default.nix#L20)), but nixpkgs wants it disabled. mkForcing the option to be disabled causes my pi to not boot, so I need to mkForce it to be enabled. But, if I do that, an internal nixpkgs option gets set for me (declaration: [`system.build.installBootLoader`](https://github.com/NixOS/nixpkgs/blob/206c278b970d6eb8c1a561536cbecc7b94be636a/nixos/modules/system/activation/activation-script.nix#L206) - set [here](https://github.com/NixOS/nixpkgs/blob/5d151d4d94415f9f75b4e039b6707af46e5ba6dc/nixos/modules/system/boot/loader/init-script/init-script.nix#L46-L50)) to a value that conflicts with one that rpi-nix changes through `initScript`. This feels like a circular dependency tree of things being broken and I don't know how to fix it.

Sorry if this reads more like a journal entry than a github issue, I'm trying to process what's going on and these are my observations.

@tstat I'm really lost with this, do you have any insight?

PowerUser64, 2024-08-27:
I'm not sure if I can salvage the boot disk, I'm going to rebuild it from scratch and copy files over. It's not that big of a deal because I've been meaning to start from a fresh nix config for a while now. Just means more downtime.

By the way, @tstat it might be a good idea to mention somewhere in the readme that you need to set `networking.networkmanager.plugins = lib.mkForce []` or else you have to build webkitgtk (in my experience). Found this using [nix-tree](https://github.com/utdemir/nix-tree).

closed 2024-08-27 by PowerUser64

PowerUser64, 2024-09-03:
UPDATE: It's working now! I'm pretty sure the problem was that I had a really old eeprom version and I didn't realize it. I'm thinking the eeprom I had wasn't compatible with booting with the extlinux options, and the removal of u-boot caused my system to break. Either way, I've updated the bootloader to the latest version following [these steps on the nixos wiki](https://wiki.nixos.org/wiki/NixOS_on_ARM/Raspberry_Pi_4#Updating_U-Boot/Firmware) and now I can boot!

I haven't tested booting with rpi-nix yet, but I'm pretty sure if I tried now it would work.

tstat, 2024-09-19:
There is a lot in this issue. It's somewhat hard to follow. The first error you posted:

> error: The option `boot.loader.generic-extlinux-compatible.enable' has conflicting definition values:
>       - In `/nix/store/4f9cx80vcs558qa1i68aa74yhc0lxsk5-source/raspberry-pi/4': true
>       - In [...]

Looks like a conflict that involves [nixos-hardware](https://github.com/NixOS/nixos-hardware/blob/f17f79c796381b93e13a6d83580be780da8f9697/raspberry-pi/4/default.nix#L35) and this repo. This repo uses the default rpi boot process by default, which uses the initScript bootloader. `nixos-hardware` used uboot last I saw, but I think maybe it is something else now. 

But yes, you don't want to set both of these. By forcing both of them to be on it seems like further problems were encountered.

> By the way, @tstat it might be a good idea to mention somewhere in the readme that you need to set networking.networkmanager.plugins = lib.mkForce [] or else you have to build webkitgtk (in my experience). Found this using [nix-tree](https://github.com/utdemir/nix-tree).

I don't use networkmanager personally. Does nixpkgs set enable some plugin that depends on webkitgtk?

Are you booting with uboot or the initscript method now?

PowerUser64, 2024-09-22:
> Looks like a conflict that involves [nixos-hardware](https://github.com/NixOS/nixos-hardware/blob/f17f79c796381b93e13a6d83580be780da8f9697/raspberry-pi/4/default.nix#L35)

Ohhh! I didn't know nixos-hardware sets that. I should've checked my other inputs. This could be another good thing to add a note about in the readme. Right now I have `boot.loader.generic-extlinux-compatible` enabled, but to be honest I'm not entirely sure what that implies. The rpi boot process (and things surrounding it like uboot, and extlinux-compatible) are black magic to me, and despite my research, I still don't know where it's documented to learn more about it. If you happen to know of anything you can point me to on this I would love to learn more about how it works! I might be able to have a more intelligent discussion about what's going on if I actually understood what I'm talking about.

> Does nixpkgs set enable some plugin that depends on webkitgtk?

Yeah, I'm pretty sure it's the openconnect plugin, which is [enabled by default](https://github.com/NixOS/nixpkgs/blob/nixos-unstable/nixos/modules/services/networking/networkmanager.nix#L635), and depends on `stoken`, which in turn depends on gtk and a bunch of other graphical things.

> Are you booting with uboot or the initscript method now?

I believe setting `boot.loader.generic-extlinux-compatible` means I'm using initscript, but I'm not 100% sure to be honest. I just tried switching back to `raspberry-pi-nix` again (and commented out my own boot settings as well as nixos-hardware) and it left me with no boot, so I'm really not sure what to make of it. (Don't worry though, I got it back and running in ~30 min by overwriting the boot partition with the one from the stock NixOS boot image [on hydra](https://hydra.nixos.org/job/nixos/trunk-combined/nixos.sd_image.aarch64-linux).)

The method I'm using is going to be either what nixos-hardware sets or what the nixos stock boot image from hydra uses.

---

Thanks for getting back to me on this! And sorry about how fragmented everything in this issue is. I was very anxious and uncertain when I was writing most of it and I tried to document as much as I could since I didn't know where the red herrings were, but I think I might've gone a bit overboard. I hope we can get to the bottom of whatever is going on! I could share a zip of my pi's config with you if you'd like. It's not public right now (it might be someday), but if I would be happy to share it with you if you think it would help figure out what's happening. Are you on matrix, IRC, or another chat platform? It might be easier to send a file privately (or a link to one) on another platform. I'm PowerUser64 in those places. I could also encrypt it with a public key of yours (gpg, [ssh](https://github.com/tstat.keys), or similar) and send it that way if you'd like.

DerRockWolf, 2024-09-23:
Regarding `webkitgtk`:

`webkitgtk` is being rebuilt because it depends on `libcamera` & `libpisp`.

Another workaround would be `raspberry-pi-nix.libcamera-overlay.enable = false;`.

@tstat would it be possible to add the overlay only to packages that need it or is the overlay meant to be applied to all packages that depend on it?

tstat, 2024-10-02:
Well, it was intended to be applied to all packages that depend on it, as I assume whatever changes were made in the fork are necessary and/or helpful if you want to use rpi hardware, but I don't know any details about that stuff.

We could just limit those packages to just be inputs to rpicam-apps, but I wouldn't be surprised if the mainline libcamera didn't work appropriately for rpi hardware with other downstream packages.