nix-community/raspberry-pi-nix #57: Warn about old eeprom versions

PowerUser64, 2024-09-03:
In order to prevent issues with booting with newer boot methods, I think rpi-nix should warn users if they have an old bootloader version (eg. <january 2023 or <1 year ago), or provide an option to update it. See also: https://github.com/nix-community/raspberry-pi-nix/issues/48#issuecomment-2325941758.

tstat, 2024-09-12:
Thanks for writing all of this up. I haven't had time to read it yet, but I intend to soon.

tstat, 2024-11-05:
IIUC (reading the linked thread) we don't know of old eeprom versions being an issue. Please reopen if I missed something.

closed 2024-11-05 by tstat

PowerUser64, 2024-11-05:
Yeah, I guess that's the conclusion here. I'll let you know if anything changes, but for now it seems like I can't use rpi-nix since my bootloader explodes with it :confused:.

tstat, 2024-11-05:
that's strange. I would be interested to know if you can't boot the example config. https://github.com/nix-community/raspberry-pi-nix/blob/master/flake.nix#L62-L64

PowerUser64, 2024-11-05:
Sure, I'll give it a test! Although it might not be until next month, since right now I don't have time to recover if I break anything :sweat_smile:.