nix-community/raspberry-pi-nix #62: libcamera URL broken

KP64, 2024-09-12:
Updating or installing the new raspberry-pi-nix version isn't possible, as the libcamera URL seems to be invalid.
[image]

tstat, 2024-09-12:
Ah, it's the `+` in the url: https://github.com/nix-community/raspberry-pi-nix/commit/0ffba0a5bfb541fc4cb3041a33d956ddd71fc512#diff-206b9ce276ab5971a2489d75eb1b12999d4bf3843b7988cbe8d687cfde61dea0R36

I think that was fixed in a more recent nix version. So, we should use the SHA instead of the tag for that pin.

KP64, 2024-09-12:
The issue has been resolved. Thank you for your help.
As you pointed out, it was the nix version that I was using causing the issue.
I upgraded from 2.18.5 to the latest (2.23.3) and now everything works perfectly fine.
Closing the issue :)

closed 2024-09-12 by KP64