nix-community/raspberry-pi-nix #71: configuration trying to build webkitgtk-2.44

Strix-Vyxlor, 2024-10-02:
I have a small problem, my configuration (see below) for building an sd-image tries to compile an older version of gtkwebkit
I'm am using nixpkgs-unstable, and the latest version of gtkwebkit is 2.46.0+abi=4.0, but it tries to compile 2.44.0.
I cannot find where this package is even required.
I would build the package myself, but when compiling on 8 cores and 32Gb ram it runs out of memory (this is on wsl tho).
And when looking on hydra there are prebuild packages available.

I am able to (remote-)build with the following devices (all devices cross compile)
- my server (old laptop): i7-4xxx 8cores, 16Gb ram, nixos x64.
- main laptop: ryzen 5 5600U 12cores, 16Gb ram, nixos x64.
- gaming laptop: i7-7xxxHK 8cores, 32Gb ram, wsl ubuntu with nix.

Zix is my own nix-helper and is very work in progress: https://github.com/Strix-Vyxlor/zix
```{
  pkgs,
  lib,
  zix-pkg,
  systemSettings,
  userSettings,
  inputs,
  ...
}: {
  imports = [
    inputs.raspberry-pi-nix.nixosModules.raspberry-pi
    ../../system/security/doas.nix
    ../../system/security/gpg.nix
  ];
  users.users.root.initialPassword = "root";

  raspberry-pi-nix.board = "bcm2712";
  raspberry-pi-nix.uboot.enable = false;

  nix.package = pkgs.nixFlakes;
  nix.extraOptions = ''
    experimental-features = nix-command flakes
  '';
  nixpkgs.config.allowUnfree = true;

  nix.settings = {
    substituters = [
      "https://nix-community.cachix.org"
    ];
    trusted-public-keys = [
      "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
    ];
  };

  boot.plymouth.enable = true;

  networking.hostName = systemSettings.hostname;
  networking.networkmanager.enable = true;

  time.timeZone = systemSettings.timezone;
  i18n.defaultLocale = systemSettings.locale;
  i18n.extraLocaleSettings = {
    LC_ADDRESS = systemSettings.locale;
    LC_IDENTIFICATION = systemSettings.locale;
    LC_MEASUREMENT = systemSettings.locale;
    LC_MONETARY = systemSettings.locale;
    LC_NAME = systemSettings.locale;
    LC_NUMERIC = systemSettings.locale;
    LC_PAPER = systemSettings.locale;
    LC_TELEPHONE = systemSettings.locale;
    LC_TIME = systemSettings.locale;
  };

  users.users.${userSettings.username} = {
    isNormalUser = true;
    description = userSettings.name;
    extraGroups = ["networkmanager" "wheel" "input" "libvirtd"];
    packages = [];
    uid = 1000;
  };

  environment.systemPackages = with pkgs; [
    zix-pkg
    helix
    wget
    git
    home-manager
    wpa_supplicant
    bluez
    bluez-tools
  ];

  fonts.fontDir.enable = true;

  console.keyMap = "be-latin1";

  system.stateVersion = "24.05";

  hardware = {
    raspberry-pi = {
      config = {
        all = {
          options = {
            # The firmware will start our u-boot binary rather than a
            # linux kernel
            arm_64bit = {
              enable = true;
              value = true;
            };
            disable_overscan = {
              enable = true;
              value = true;
            };
          };
          dt-overlays = {
            vc4-kms-v3d = {
              enable = true;
              params = {};
            };
          };
        };
      };
    };
  };
}```

Sorry for any typos or bad English, I'm dyslectic and non native English speaker.

tstat, 2024-10-02:
Hey @Strix-Vyxlor. Yes, this issue has been mentioned here https://github.com/nix-community/raspberry-pi-nix/issues/48#issuecomment-2369426119, and the dependency reportedly arises from a networkmanager plugin.

If you don't care about using the rpi libcamera fork, then the easiest thing to do would be to disable it by setting the `libcamera-overlay.enable` setting to `false` ([src](https://github.com/nix-community/raspberry-pi-nix/blob/master/rpi/default.nix#L46))

Strix-Vyxlor, 2024-10-03:
Now I have a different issue, it fails to build linux-6.10.8-modules-shrunk (same with v6.6.47).
```error: builder for '/nix/store/33dj7wrcpv2v658p1im0yw0b4w9szrxc-linux-6.10.8-modules-shrunk.drv' failed with exit code 1;
       last 25 log lines:
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/media/cec/core/cec.ko.xz
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/gpu/drm/display/drm_display_helper.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/core/snd.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/core/snd-timer.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/core/snd-pcm.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/core/snd-pcm-dmaengine.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/ac97_bus.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/core/snd-compress.ko.xz
       >   dependency already copied: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/soc/snd-soc-core.ko.xz
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/sound/soc/codecs/snd-soc-hdmi-codec.ko.xz
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/gpu/drm/vc4/vc4.ko.xz
       > root module: pcie_brcmstb
       >   builtin dependency: pcie_brcmstb
       > root module: reset-raspberrypi
       >   builtin dependency: reset_raspberrypi
       > root module: ext2
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/fs/ext2/ext2.ko.xz
       > root module: ext4
       >   builtin dependency: ext4
       > root module: tpm-tis
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/char/tpm/tpm.ko.xz
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/char/tpm/tpm_tis_core.ko.xz
       >   copying dependency: /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8/kernel/drivers/char/tpm/tpm_tis.ko.xz
       > root module: tpm-crb
       > modprobe: FATAL: Module tpm-crb not found in directory /nix/store/g14qg8h575r84hv1mrigha48zfqfwadq-linux-6.10.8-modules/lib/modules/6.10.8
       For full logs, run 'nix log /nix/store/33dj7wrcpv2v658p1im0yw0b4w9szrxc-linux-6.10.8-modules-shrunk.drv'.
error: 1 dependencies of derivation '/nix/store/46ghx06a1cvfv9qp7zvvnppklb1sw4zh-initrd-linux-6.10.8.drv' failed to build
error: 1 dependencies of derivation '/nix/store/6464kiq1f1n9v3dj8i6v37cma4ayhkc8-nixos-system-raspberrypi-24.11.20240929.06cf0e1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/rwx27103ld983rwhg8zwn7ki0cc6b7pr-ext4-fs.img.zst.drv' failed to build
error: 1 dependencies of derivation '/nix/store/96gx993qpwgs2ks1yc219v6msjqkrjf7-nixos-sd-image-24.11.20240929.06cf0e1-aarch64-linux.img.drv' failed to build```

closed 2024-10-03 by Strix-Vyxlor