nix-community/raspberry-pi-nix #89: Unable to nixos-rebuild switch with PR79 when doing flake update

victorbiga, 2024-11-15:
https://github.com/nix-community/raspberry-pi-nix/pull/79/files update broke nixos-rebuild switch when flake is updated, affecting RPI5

My config is public https://github.com/victorbiga/k3s-nixos-pi

Any guidance would be appreciated on how to upgrade existing running system.

Here are the logs:

```
building the system configuration...
warning: Git tree '/etc/nixos' is dirty
error:
       … while calling anonymous lambda
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/attrsets.nix:1570:24:
         1569|     let f = attrPath:
         1570|       zipAttrsWith (n: values:
             |                        ^
         1571|         let here = attrPath ++ [n]; in

       … while calling the 'head' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/attrsets.nix:1574:11:
         1573|         || pred here (elemAt values 1) (head values) then
         1574|           head values
             |           ^
         1575|         else

       … while calling anonymous lambda
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/attrsets.nix:1204:18:
         1203|         mapAttrs
         1204|           (name: value:
             |                  ^
         1205|             if isAttrs value && cond value

       … from call site
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/attrsets.nix:1207:18:
         1206|             then recurse (path ++ [ name ]) value
         1207|             else f (path ++ [ name ]) value);
             |                  ^
         1208|     in

       … while calling anonymous lambda
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:254:72:
          253|           # For definitions that have an associated option
          254|           declaredConfig = mapAttrsRecursiveCond (v: ! isOption v) (_: v: v.value) options;
             |                                                                        ^
          255|

       … while evaluating the attribute 'value'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:816:9:
          815|     in warnDeprecation opt //
          816|       { value = addErrorContext "while evaluating the option `${showOption loc}':" value;
             |         ^
          817|         inherit (res.defsFinal') highestPrio;

       … while evaluating the option `system.build.toplevel':

       … while evaluating the attribute 'mergedValue'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:851:5:
          850|     # Type-check the remaining definitions, and merge them. Or throw if no definitions.
          851|     mergedValue =
             |     ^
          852|       if isDefined then

       … while evaluating a branch condition
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:852:7:
          851|     mergedValue =
          852|       if isDefined then
             |       ^
          853|         if all (def: type.check def.value) defsFinal then type.merge loc defsFinal

       … while evaluating the attribute 'values'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:845:9:
          844|       in {
          845|         values = defs''';
             |         ^
          846|         inherit (defs'') highestPrio;

       … while evaluating a branch condition
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:841:11:
          840|           # Avoid sorting if we don't have to.
          841|           if any (def: def.value._type or "" == "order") defs''.values
             |           ^
          842|           then sortProperties defs''.values

       … while calling the 'any' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:841:14:
          840|           # Avoid sorting if we don't have to.
          841|           if any (def: def.value._type or "" == "order") defs''.values
             |              ^
          842|           then sortProperties defs''.values

       … while evaluating the attribute 'values'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:944:7:
          943|     in {
          944|       values = concatMap (def: if getPrio def == highestPrio then [(strip def)] else []) defs;
             |       ^
          945|       inherit highestPrio;

       … while calling the 'concatMap' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:944:16:
          943|     in {
          944|       values = concatMap (def: if getPrio def == highestPrio then [(strip def)] else []) defs;
             |                ^
          945|       inherit highestPrio;

       … while calling the 'concatMap' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:831:17:
          830|         # Process mkMerge and mkIf properties.
          831|         defs' = concatMap (m:
             |                 ^
          832|           map (value: { inherit (m) file; inherit value; }) (addErrorContext "while evaluating definitions from `${m.file}':" (dischargeProperties m.value))

       … while calling anonymous lambda
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:831:28:
          830|         # Process mkMerge and mkIf properties.
          831|         defs' = concatMap (m:
             |                            ^
          832|           map (value: { inherit (m) file; inherit value; }) (addErrorContext "while evaluating definitions from `${m.file}':" (dischargeProperties m.value))

       … while calling the 'map' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:832:11:
          831|         defs' = concatMap (m:
          832|           map (value: { inherit (m) file; inherit value; }) (addErrorContext "while evaluating definitions from `${m.file}':" (dischargeProperties m.value))
             |           ^
          833|         ) defs;

       … while evaluating definitions from `/nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/nixos/modules/system/activation/top-level.nix':

       … from call site
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:832:128:
          831|         defs' = concatMap (m:
          832|           map (value: { inherit (m) file; inherit value; }) (addErrorContext "while evaluating definitions from `${m.file}':" (dischargeProperties m.value))
             |                                                                                                                                ^
          833|         ) defs;

       … while calling 'dischargeProperties'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:903:25:
          902|   */
          903|   dischargeProperties = def:
             |                         ^
          904|     if def._type or "" == "merge" then

       … while evaluating a branch condition
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:904:5:
          903|   dischargeProperties = def:
          904|     if def._type or "" == "merge" then
             |     ^
          905|       concatMap dischargeProperties def.contents

       … while evaluating the attribute 'value'
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/lib/modules.nix:614:53:
          613|                 (n: value:
          614|                   [{ inherit (module) file; inherit value; }]
             |                                                     ^
          615|                 )

       … while calling the 'throw' builtin
         at /nix/store/5fd8pq6w9i7brdhwhpj0pbgg4s3bk5v5-source/nixos/modules/system/activation/top-level.nix:73:10:
           72|   baseSystemAssertWarn = if failedAssertions != []
           73|     then throw "\nFailed assertions:\n${concatStringsSep "\n" (map (x: "- ${x}") failedAssertions)}"
             |          ^
           74|     else showWarnings config.warnings baseSystem;

       error:
       Failed assertions:
       - The ‘fileSystems’ option does not specify your root file system.
     ```

orzklv, 2024-11-17:
@victorbiga  the issue has been resolved by adding sd-image module like this:

[image]

tstat, 2024-11-17:
Yes, you can import the sd-image module to retain the old behavior. Check out the readme and example build.

That being said. I don't encourage tracking the development branch. If you want to avoid breaking changes you should track releases.

closed 2024-11-17 by tstat

orzklv, 2024-11-18:
@tstat Agreed, I guess adding a warning for new comers on readme would be sufficient. Like:

**Use release track/branch to avoid unexpected issues. If you're going to use master, you should refer to source code and keep up with developers on your own.**

tstat, 2024-11-18:
https://github.com/nix-community/raspberry-pi-nix/commit/aaec735faf81ff05356d65c7408136d2c1522d34