nix-community/raspberry-pi-nix #95: Kernel 6.6.54 not in Cachix

giomf, 2024-12-04:
Hello everyone,

I'm using your module for the first time and I'm not sure if I've understood it correctly but it seems that the kernel 6.6.54 is not in Cachix so building it for my pi4 takes ages.

When executing nixos-rebuild it says that the following packages will be build: 

```
these 10 derivations will be built:
  /nix/store/lyjhn00y9a0011gxishq7ysmycqqmljj-linux-6.6.54.drv
  /nix/store/yxc04l3hhv6pq8256cizvxsq7hd24jvq-migrate-rpi-firmware.drv
  /nix/store/ycjlqqwgww50asjqzyhm5yj5fabcnw01-unit-raspberry-pi-firmware-migrate.service.drv
  /nix/store/nwj6nlc7cbrpz7k8nc7vlzirza1p97b3-system-units.drv
  /nix/store/1iqrnv9pgnsfzs9f5inpyqn2i1ajppvz-etc.drv
  /nix/store/46c0x64f2b5s59151yvz0zkv527227fv-linux-6.6.54-modules.drv
  /nix/store/n22icqmwkwfccshvw869grphxc62ifs6-linux-6.6.54-modules-shrunk.drv
  /nix/store/zmcnsxr8zrg991kpzyi11zffx9fy7j3c-initrd-linux-6.6.54.drv
  /nix/store/w83l6jjkbz2yf0w2yc76xzb60a74rabj-boot.json.drv
  /nix/store/0xw2r33rw3q907nm4dhns5fb357a26xn-nixos-system-homeassistant-24.05.20241201.7e1ca67.drv
```

First I thought that I messed up the cachix configuration but it seems that the package is not present:

```
curl https://nix-community.cachix.org/lyjhn00y9a0011gxishq7ysmycqqmljj.narinfo
404 - Not Found
```

So, did I something wrong or is the kernel version simply not in cachix?

Infinidoge, 2024-12-07:
I've experienced this same issue. It's annoying enough that I just `lib.mkForce`'d it to be the Nixpkgs rpi 3/4 kernel.

(Speaking of which, why isn't there a toggle option to use the nixpkgs kernel?)

giomf, 2024-12-07:
That's sad indeed.

I have some stability issues and hoped that the custom kernel would solve it.

Does anybody know if and how I could request the package to be added to cachix?

giomf, 2024-12-10:
> I've experienced this same issue. It's annoying enough that I just lib.mkForce'd it to be the Nixpkgs rpi 3/4 kernel.

@Infinidoge Could you tell me how you manage to do it please?

Infinidoge, 2024-12-10:
```
boot.kernelPackages = lib.mkForce pkgs.linuxKernel.packages.linux_rpi4;
```
(Or `rpi3`, of course)

giomf, 2024-12-11:
Thanks!

That was easier then I thought.

adminy, 2024-12-14:
I build the kernel on the mac / pc then I deploy it pre-built to the pi.

giomf, 2024-12-18:
Yes of course this works. But for my relatively old machine it take really long.
Also since, its mentioned in the readme that cachix should work, I would consider this a bug.

adminy, 2024-12-20:
which kernel are you using, I'll test it to see if its cached or not on my end. It could just be that your cachix key or url is not configured in your nix.conf file.

zeorin, 2024-12-22:
I also have this issue. I'm using flakes. I have the following in my flake.nix:

```nix
{
  nixConfig = {
    extra-substituters = [
      "https://nix-community.cachix.org"
      "https://raspberry-pi-nix.cachix.org"
    ];
    extra-trusted-public-keys = [
      "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
      "raspberry-pi-nix.cachix.org-1:WmV2rdSangxW0rZjY/tBvBDSaNFQ3DyEQsVw8EvHn9o="
    ];
  };
}
```

giomf, 2024-12-22:
> which kernel are you using, I'll test it to see if its cached or not on my end. It could just be that your cachix key or url is not configured in your nix.conf file.

Im using version 6.6.54. Also cachix is setup correct. I can see how its been used for some packages.

Using `curl https://nix-community.cachix.org/lyjhn00y9a0011gxishq7ysmycqqmljj.narinfo` shows that the kernel is not in cachix.

zeorin, 2024-12-26:
BTW seems kernel 6.10.12 is also not in the cache.

tstat, 2025-01-08:
The kernels built by CI are pushed into the community cachix. For example:

```
# nix path-info .#linux-v6_6_67-bcm2711
/nix/store/1jw3h4jspvapmzjqykjcbygw7s4pi47c-linux-rpi-6.6.67

# curl https://nix-community.cachix.org/1jw3h4jspvapmzjqykjcbygw7s4pi47c.narinfo
StorePath: /nix/store/1jw3h4jspvapmzjqykjcbygw7s4pi47c-linux-rpi-6.6.67
URL: nar/41d3b501426227b90118a72fe9434e92102f8c654d27e079de01ed6407e03544.nar.zst
Compression: zstd
FileHash: sha256:41d3b501426227b90118a72fe9434e92102f8c654d27e079de01ed6407e03544
FileSize: 110335082
NarHash: sha256:0hc6bkmahr6ys8hi67yv77b178gbf36y00g18ymifg5fr2l0lm2j
NarSize: 143699312
References: 1jw3h4jspvapmzjqykjcbygw7s4pi47c-linux-rpi-6.6.67
Deriver: 161ry7bbjymrw6pd620iprz4wrly0g7b-linux-rpi-6.6.67.drv
Sig: nix-community.cachix.org-1:3kSE0BgnptQ61L6/XPFpLTC8FWmk9oQ0/IPjCHOujol5pMnCshad6dSOVaFfQp79nBO2q8vhFC2CvFz2TcHACA==
```

So any cache misses are likely due to having different inputs. You should compare what kernel CI built with something like

```
nix derivation show .#linux-v6_6_67-bcm2711
```

in raspberry-pi-nix

to what kernel you are trying to build in your configuration, with something like

```
nix derivation show .#nixosConfigurations.<YOUR_CONFIG_NAME>.config.system.build.kernel
```

tstat, 2025-01-08:
Closing, feel free to reopen with the appropriate info.

closed 2025-01-08 by tstat

tennox, 2025-01-09:
I've tried what you said, no diff.

<details><pre>
❯ nix derivation show .#nixosConfigurations.rpi-example.config.system.build.kernel > my-kernel.json
evaluating derivation 'git+file:///home/manu/dev/raspi/raspi-nix#nixosConfigurations.rpi-example.config.systemevaluating derivation 'git+file:///home/manu/dev/raspi/raspi-nix#nixosConfigurations.rpi-example.config.system
raspi-nix on  master [$!?] took 2s
❯ nix derivation show .#packages.aarch64-linux.linux-v6_6_51-bcm2711 > pkg-kernel.json

raspi-nix on  master [$!?]
❯ difft *kernel.json
pkg-kernel.json --- JSON
No changes.
</pre></details>

But in the end I figured there was a recent kernel change: https://github.com/nix-community/raspberry-pi-nix/pull/107
& the public key of the cachix changed - which in the end caused my machine trying to build a linux kernel for hours.

Pushed this as my suggested fix: https://github.com/nix-community/raspberry-pi-nix/

zeorin, 2025-01-26:
I'm building for RPi5, so `board = "bcm2712"`