nix-community/raspberry-pi-nix #115: Issue with new kernel.moduleMakeFlags flags missing

koalalorenzo, 2025-01-28:
Hello!
First of all: I am very happy that this repository exists because so far it is the only way for me to get my Raspberry Pi 5 to work with nix. Thank you for it!

I am upgrading now my NAS from a raspberry pi 4 to a brand new raspberry pi 5. The NAS uses ZFS.

When I enable ZFS and use this repo to build an sd image I encountered this error:

```
$ nix build ".#nixosConfigurations.pi5_nas.config.system.build.sdImage" --impure --accept-flake-config --extra-experimental-features flakes --extra-experimental-features nix-command

warning: Git tree '/mnt/share/Repositories/nixos-config' is dirty
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:34:12:
           33|
           34|   strict = derivationStrict drvAttrs;
             |            ^
           35|

       … while evaluating derivation 'nixos-sd-image-24.11.20250126.4e96537-aarch64-linux.img'
         whose name attribute is located at /nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/pkgs/stdenv/generic/make-derivation.nix:336:7

       … while evaluating attribute 'buildCommand' of derivation 'nixos-sd-image-24.11.20250126.4e96537-aarch64-linux.img'
         at /nix/store/mwcv486c408zz6gvm0qzdjmmy7sdmhfq-source/sd-image/sd-image.nix:177:11:
          176|
          177|           buildCommand = ''
             |           ^
          178|             mkdir -p $out/nix-support $out/sd-image

       … while evaluating the option `sdImage.populateRootCommands':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/flake.nix':

       … while evaluating the option `system.build.toplevel':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/nixos/modules/system/activation/top-level.nix':

       … while evaluating the option `system.systemBuilderArgs':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/nixos/modules/system/activation/activatable-system.nix':

       … while evaluating the option `system.activationScripts.etc.text':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/nixos/modules/system/etc/etc-activation.nix':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/nixos/modules/system/etc/etc.nix':

       … while evaluating the option `environment.etc."systemd/system".source':

       … while evaluating the option `systemd.units."raspberry-pi-firmware-migrate.service".text':

       … while evaluating definitions from `/nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/nixos/modules/system/boot/systemd.nix':

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: attribute 'moduleMakeFlags' missing
       at /nix/store/50yickar04m51aqnc43gxf45g2i0n3k9-source/pkgs/os-specific/linux/zfs/generic.nix:212:14:
          211|           ]
          212|           ++ kernel.moduleMakeFlags
             |              ^
          213|         );
make[1]: *** [Makefile:11: build_nixosConfigurations.pi5_nas.config.system.build.sdImage] Error 1
make[1]: Leaving directory '/mnt/share/Repositories/nixos-config'
make: *** [Makefile:23: sd_pi5_nas] Error 2
```

Where my `pi5_nas` configuration has these parameters changed:

```nix
{ ... } : {
  raspberry-pi-nix.board = "bcm2712";
  hardware.raspberry-pi.config.all.options = {
    fan_temp0 = {
      enable = true;
      value = 0;
    };

    fan_temp0_speed = {
      enable = true;
      value = 224;
    };
  };
  raspberry-pi-nix = {
    kernel-version = "v6_10_12";

    libcamera-overlay.enable = false;
  };
 
  fileSystems."/".options = [ "noatime" ];
  boot.initrd.availableKernelModules = [ "xhci_pci" "usbhid" "usb_storage" ];

  # console.enable = false;
  environment.systemPackages = with pkgs; [
    libraspberrypi
    raspberrypi-eeprom
  ];
  hardware.bluetooth.enable = false;

}
```

koalalorenzo, 2025-01-29:
I had to dig a little deeper.

It seems that the issue is related to this configuration:

```nix
boot.supportedFilesystems = [ "zfs" "ext4" "ntfs" "btrfs" ];
```

Which I assume will try to build the kernel module for zfs. I wonder if the raspberry-pi kernel needs to be rebuilt or something, in order to fill the missing `kernel.moduleMakeFlags`.

I hope this info helps.

mroi, 2025-01-29:
I am not sure if this is related to the issue discussed here, but it might be. I am using the following construction in my config:

```
nixpkgs.overlays = [ (final: prev: {
	makeModulesClosure = x: prev.makeModulesClosure (x // { allowMissing = true; });
})];
```

This prevents the kernel build from complaining when modules are missing.

koalalorenzo, 2025-01-29:
> I am not sure if this is related to the issue discussed here, but it might be. I am using the following construction in my config:
>
> ```
> nixpkgs.overlays = [ (final: prev: {
> 	makeModulesClosure = x: [...]

@mroi I tried this code, as well as changing it to other values, but I am not sure how to use it in order to make `kernel.moduleMakeFlags` or `zfs` to allow missing when building 🤔

Any suggestions?

mroi, 2025-01-29:
I just put this in my NixOS `configuration.nix` because the build of the kernel modules would otherwise complain about missing modules. I do not recall if it was about ZFS or something else though…

koalalorenzo, 2025-02-04:
That workaround is not working for me (I spent too much time trying to make it work 😅), but thank you for suggesting @mroi :)

mroi, 2025-02-04:
Sorry for the noise then. It may be useful for a different issue…

koalalorenzo, 2025-02-11:
It is not a proper fix, but I manage to work around by recompiling the kernel and forcing my flake input to map the nixpkgs of raspberry-pi-nix to the current version I am using in nixpkgs:

```nix
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.11";
    raspberry-pi-nix.url = "github:nix-community/raspberry-pi-nix/v0.4.1";
    raspberry-pi-nix.inputs.nixpkgs.follows = "nixpkgs";
  };
```

This is because I did not want to stay out-of-date with nixos-24.11 release, and use the latest stable raspberry-pi-nix. I tried to use the master branch of raspberry-pi-nix, but I had other issues like in #120 with the Ethernet not working (with all Linux kernel versions available, including v6.12). It took me a few days.

@tstat do you know what could have been the issue? Should I open a new issue in nixpkgs repo too?