nix-community/raspberry-pi-nix #116: Define a kernel policy

JamieMagee, 2025-01-29:
I touched on this a little bit in my recent kernel update PRs (#101 and #102), but this repository doesn't have a policy about which kernels are defined in `flake.nix`. Currently we have:

- `6.6.51`
- `6.6.67`
- `6.10.12`

I think a good policy might be something like:

- Latest `stable_*` tag
- Latest LTS kernel release 
  - Currently `rpi-6.12.y`
    - Either the tag `rpi-6.12.y_20241206_2` which is `6.12.1`
    - Or the tip of `rpi-6.12.y` which is currently `6.12.11`
- Latest mainline kernel release
  - Currently `rpi-6.13.y`

tstat, 2025-01-30:
Yes, good idea. I also think that's a good policy. I also think defaulting the the latest stable (as is done currently) is best.

adminy, 2025-02-08:
Honestly I'm more excited to adopt the mainline kernel so that it aligns with whatever its in nixpkgs.

pta2002, 2025-02-11:
Why not mirror what NixOS does? https://wiki.nixos.org/wiki/Linux_kernel
- Default is latest LTS
- Also available will be:
  - For stable: latest kernel version at the time of stabilization
  - For unstable: latest kernel version

JamieMagee, 2025-02-20:
Unfortunately, I don't think upstream NixOS policy maps as well onto the Raspberry Pi.

Setting the default kernel as the latest Linux LTS release would be `6.12.15`, which is _not_ tagged in `raspberrypi/linux`. Whereas the newest `stable_` tag from [`raspberrypi/linux`][1] is currently `stable_20250127` which is [`6.6.74`][2], and the latest published kernel for Raspberry Pi OS is `6.6.51`[^1] (AKA the `stable_20241008` tag).

I'd advocate for tracking what is shipped to Raspberry Pi OS over the current Linux LTS version.

[^1]: https://downloads.raspberrypi.com/raspios_arm64/release_notes.txt

[1]: https://github.com/raspberrypi/linux/tags
[2]: https://github.com/raspberrypi/linux/blob/a18d9ced4965462cb7b3b4252ada440395105308/Makefile#L2-L4

pta2002, 2025-02-20:
Hadn't considered that, but it is a good point. Still, I think it makes sense to always provide whatever the latest release by rpi is, since it's not always necessarily the primary one they release. For instance they were on 6.6 officially for a good while but the patched sources for later versions were still available.