nix-community/raspberry-pi-nix #120: Network deivce type "bridge" unknown after update

Svenum, 2025-02-09:
After updating my flake `nix flake update` my raspberry pi 5 is not able to create a bridge device "br0".

I get this error:

```bash
ip link add name br0 type bridge
Error: Unknown device type
```

Any idea?

Here is my config: https://github.com/Svenum/holynix/blob/main/systems/aarch64-linux/srv-raspi5/default.nix

koalalorenzo, 2025-02-11:
I have noticed a similar issue. After upgrading to test latest kernels, my rpi Ethernet stopped working. Downgrading to a different kernel did not fix the issue. I am still investigating, and upgrading the firmware, but something feels off. Using a USB Ethernet dongle works for now, but it is a workaround 😅

Sticking to v0.4.1 branch in the flake config fixed the issue for me!

pta2002, 2025-02-11:
Using the firmware files from the [next branch](https://github.com/raspberrypi/firmware/tree/next/boot) worked for me. Perhaps it is best if `raspberry-pi-nix` starts building these by itself like the upstream `linux_rpi4` kernel does? That way they won't ever go out of sync.

Reference: https://github.com/NixOS/nixpkgs/blob/a45fa362d887f4d4a7157d95c28ca9ce2899b70e/pkgs/os-specific/linux/kernel/linux-rpi.nix#L96-L135